pvlib/pvlib-python #1887: `lambertw` is slowed down by unnecessary pandas operations in `_golden_sect_DataFrame`

kandersolar, 2023-10-10:
**Is your feature request related to a problem? Please describe.**
Specifying `method='lambertw'` when solving the SDE makes a call to `tools._golden_sect_DataFrame` to locate the maximum power point.  `_golden_sect_DataFrame` is a vectorized implementation of the [golden-section search](https://en.wikipedia.org/wiki/Golden-section_search), an iterative algorithm for locating the maximum (or minimum) of a function.  It turns out that the machinery of `_golden_sect_DataFrame` (not counting the lambertw function evaluations within it) is a substantial contributor to the overall runtime of `pvlib.singlediode._lambertw`.

**Describe the solution you'd like**

`_golden_sect_DataFrame`, as the name suggests, makes heavy used of pandas objects internally.  In particular, it repeatedly (re)assigns columns into a DataFrame.  This is rather slow, and unnecessary as far as I can tell.

I observe significant speed ups in a reimplementation of the search algorithm that uses normal variables instead of a dataframe-based work space.  Here is a timing comparison of `pvlib.singlediode._lambertw`, showing 20-30% speed improvement for large inputs:

[image]

**Describe alternatives you've considered**

The golden-section search can in principle be further optimized by re-using function evaluations from the previous iteration.  I briefly looked into applying this to our code but decided it would be too much of a pain to incorporate into this vectorized implementation.

Of course ideally we would not be in the business of implementing optimization algorithms at all.  Hopefully in the future we can remove this function altogether, either by replacing it with some suitable offering from scipy or by removing the need for such an optimization in the first place.

mikofski, 2023-10-11:
Amen! Any chance a cythonized version of BrentQ could be useful?
https://gist.github.com/mikofski/ac30065073d0d32d6ea3569f6e24e5ec

kandersolar, 2023-10-11:
After further investigation, I think the speedup was actually due to a bug in my code.  After fixing it, I don't observe a meaningful difference.

I also notice that the supposed explanation of the speedup (that pandas operations are slow) is not relevant here, since the function is actually called with a dict, not a dataframe: https://github.com/pvlib/pvlib-python/blob/v0.10.2/pvlib/singlediode.py#L774-L778

Oops!  I'll close this.

closed 2023-10-11 by kandersolar