pvlib/pvlib-python #1892: Various pandas deprecation warnings from test suite

kandersolar, 2023-10-17:
We've started getting several types of deprecation warnings from the tests:

```
pvlib/tests/test_modelchain.py::test_aoi_model_interp
  /home/runner/work/pvlib-python/pvlib-python/pvlib/tests/test_modelchain.py:1476: FutureWarning: Series.__getitem__ treating keys as positions is deprecated. In a future version, integer keys will always be treated as labels (consistent with DataFrame behavior). To access a value by position, use `ser.iloc[pos]`
    assert mc.results.ac[1] < 1
```

```
pvlib/tests/test_clearsky.py::test_detect_clearsky_window
  /home/runner/work/pvlib-python/pvlib-python/pvlib/tests/test_clearsky.py:634: FutureWarning: Setting an item of incompatible dtype is deprecated and will raise in a future error of pandas. Value 'True' has dtype incompatible with int64, please explicitly cast to a compatible dtype first.
    expected.iloc[-3:] = True
```

```
pvlib/tests/test_singlediode.py: 14 warnings
  /home/runner/work/pvlib-python/pvlib-python/pvlib/tests/test_singlediode.py:118: FutureWarning: DataFrame.applymap has been deprecated. Use DataFrame.map instead.
    joined[is_number] = joined[is_number].applymap(np.float64)
```

```
  /home/runner/work/pvlib-python/pvlib-python/pvlib/scaling.py:279: FutureWarning: DataFrame.fillna with 'method' is deprecated and will raise in a future version. Use obj.ffill() or obj.bfill() instead.
    df = df.fillna(method='bfill').fillna(method='ffill')
```

Here's an example run: https://github.com/pvlib/pvlib-python/actions/runs/6550206298/job/17788805508?pr=1827#step:9:65

Addressing these would be good first PRs for anyone wanting to get started with contributing to pvlib's codebase!  If you plan to submit a PR for one of these warnings, please comment here so that someone else does not duplicate the work.

matsuobasho, 2023-10-24:
Is the intention to resolve these warnings by updating the deprecated methods with current ones, or rather to just hide the deprecation warnings?

cwhanse, 2023-10-24:
@matsuobasho preferably update the methods to resolve the warnings. If updating breaks other code, perhaps hide them.

matsuobasho, 2023-10-24:
Any recommendations on putting a breakpoint in the tests where the warnings occur in order to help with diagnosing the issue and testing for a resolution?  I see one way mentioned in this [SO](https://stackoverflow.com/questions/35988404/is-there-a-pytest-command-line-option-plug-in-for-on-demand-debugging-pdb) post, but since I'm in an environment pdb doesn't reocognize pytest when I run `python -mpdb $(which pytest)`

matsuobasho, 2023-10-28:
Ok, I'm tackling this issue.  Since is this my first issue, would like to ask about a nuance that I've encountered.
In the `test_bird` function of `test_clearsky.py`, we get a warning on lines [857-859](https://github.com/pvlib/pvlib-python/blob/main/pvlib/tests/test_clearsky.py#L857C5-L859C6):
```
irrads3 = clearsky.bird(
        zenith[12], airmass[12], aod_380nm, aod_500nm, h2o_cm, dni_extra=etr[12]
    )
```
Here's the warning:
```
FutureWarning: Series.__getitem__ treating keys as positions is deprecated. In a future version, integer keys will always be treated as labels (consistent with DataFrame behavior). To access a value by position, use `ser.iloc[pos]`
```

Now the interesting part is that that warning only pertains to `etr[12]`.  The reason why is that out of the 3 objects here that are being referenced by an index, only `etr` is a Pandas series.  `zenith` is a Pandas index type, while `airmass` is a numpy array.  Neither of those have an `iloc` method.  

So changing to `etr.iloc[12]` gets rid of the warning.  However, there is now an inconsistency in how we would be referencing these objects.  I think it's fine but looking for confirmation especially as I'll potentially run into the same scenario in other warnings that are part of this issue.  Thanks.

kandersolar, 2023-10-30:
Wow, three different vector types going into a single function :P Don't worry about the inconsistency with not using `iloc` for all inputs in that case; changing to just `etr.iloc[12]` is fine.  Maybe that test could be cleaned up a bit more generally, but that would be a separate PR.  Thanks @matsuobasho!

kandersolar, 2023-10-30:
Ah yeah, the SRML website has been down for a while and causing the associated tests to fail.  If you check the recent CI test results for `main` you'll see the same failures there.  

I suggest ignoring those failing tests for now.  If any deprecation warnings come out of those tests whenever they eventually start passing again, we can just address them at that time.

cwhanse, 2023-10-30:
@matsuobasho we add the labels to the PR.

PR #1900 (2023-11-09): Address pandas FutureWarnings in test suite

 - [ ] Closes #1892 
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added - NA since issue is about getting rid of warnings in tests
 - [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes. - NA see above
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/ [... cut]
closed 2023-11-09 by kandersolar