pvlib/pvlib-python #1894: Plotting issue with IV curve gallery example?

williamhobbs, 2023-10-20:
**Describe the bug**
When I run the IV curve example (https://pvlib-python.readthedocs.io/en/stable/gallery/iv-modeling/plot_singlediode.html), the annotation arrows show up in a second figure:

[image]

If I move `plt.show()` from the current line down below the `draw_arrow` functions, it looks correct:
[image]

Is this a bug? Or maybe there's something different with my setup?

kandersolar, 2023-10-20:
I see no reason to call `plt.show()` before finishing the figure assembly.  Its current placement in the code was probably an oversight. A PR to fix it would be welcome!

williamhobbs, 2023-10-20:
PR submitted!