pvlib/pvlib-python #1899: Package instructions for installing PVlib as editable library (possibly) incomplete

matsuobasho, 2023-10-28:
**Is your feature request related to a problem? Please describe.**
I installed pvlib as an editable library as per the [instructions](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/user_guide/installation.rst), but I did it using a `pipenv` environment.  Things work, however, I encountered some error while running pytest

```
 def test_detect_clearsky_calls_threshold(mocker, detect_clearsky_threshold_data):
E      fixture 'mocker' not found
```

```
 def test_get_pvgis_hourly(requests_mock, testfile, expected_name, args,
E      fixture 'requests_mock' not found
```
These errors went away after I installed `pytest-mock` and `requests-mock`.  

**Describe the solution you'd like**
It's possible that since the installation instructions are for conda, that conda's install of pytest also includes `pytest-mock` and `requests-mock`.  Either way it would be useful to at least mention in the setup steps that these packages also need to be installed if one is installing pvlib through another environment manager.  Or maybe these packages are required for the conda install, as well.

kandersolar, 2023-10-30:
I don't think the package manager makes a difference here.  What you've found is that `pip install pvlib` (in any environment) installs the base dependencies, but not the optional ones like the packages necessary to run the tests and build the docs.  For that, you need to run something like `pip install pvlib[test]`.  The options are specified in setup.py: 

https://github.com/pvlib/pvlib-python/blob/63a2ca47badf692cd650883f21e07c072145876e/setup.py#L49-L58

`[all]` is mentioned in the [instructions](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/user_guide/installation.rst#install-the-source-code), but maybe it should be emphasized more?

matsuobasho, 2023-10-30:
I think it's clear enough in the documentation, I missed that.  Thanks!

closed 2023-10-30 by matsuobasho