pvlib/pvlib-python #1902: Minor GTI-DIRINT bugs

adriesse, 2023-11-04:
**Describe the bug**
```gti_dirint()``` calculates a few things internally that are either not quite right or inconvenient.

1. ```airmas``` is pressure-adjusted for use in DIRINT, but Perez transposition actually needs relative airmass.
2. The airmass model is hard-coded as "kasten1966" for compatibility with DISC and DIRINT, whereas other calculations tend to use "'kastenyoung1989" as a default and the Perez model coefficients were developed using the latter.
3. ```dni_extra``` is calculated using the hard-coded constant 1370 for compatibility with DISC.  

These internal calculation make it harder to do a precise round-trip comparison between forward and reverse transposition and/or restrict the conditions under which a precise comparison is possible.

**Work-around**

1. Use ```pressure=101325```.
2. Pre-calculate ```airmass``` and ```dni_extra``` for use in the forward transposition with the internal parameters of ```gti_dirint```.

**Versions:**
 - ``pvlib.__version__``: 0.9.1

prady0t, 2023-11-04:
Hey can I try to solve this?

kandersolar, 2023-11-06:
Are those errors in our implementation (i.e. we deviate from the reference), or conceptual inconsistencies in the Marion reference itself?  If the latter, maybe we should enable these improvements via optional parameters while leaving the current behavior the default. 

@prady0t thanks for offering to help!  I think we still need to define what _should_ be done here before anyone can start doing it :)

adriesse, 2023-11-21:
I'll leave that open for discussion.  We tried hard to replicate Marion's published results but couldn't.