pvlib/pvlib-python #1908: JOSS-review: statement of need

EwaGomez, 2023-11-13:
While working through the [JOSS publication](https://github.com/openjournals/joss-reviews/issues/5994) checklist, I was looking for a statement of need for pvlib in the documentation. You mention the statement of need in the package overview introduction, but I think it would be interesting to clarify the target audience on the main landing page of the documentation as well.

kandersolar, 2023-11-15:
That landing page text is long overdue for some edits :)

#1913 goes some way towards addressing this issue.  Please let me know if you think more clarification would be helpful!

EwaGomez, 2023-11-24:
Yes, it does! I will finish my review now :-) Thanks!

closed 2023-11-24 by EwaGomez