pvlib/pvlib-python #1914: ValueError on clear-sky models page in Users Guide

kandersolar, 2023-11-15:
The [clear-sky](https://pvlib-python.readthedocs.io/en/stable/user_guide/clearsky.html#detect-clearsky) docs page has an issue: 

```python
In [156]: clear_samples = clearsky.detect_clearsky(ghi, cs['ghi'], cs.index, 10)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-156-a86cb0f5cb3e> in <module>
----> 1 clear_samples = clearsky.detect_clearsky(ghi, cs['ghi'], cs.index, 10)

~/checkouts/readthedocs.org/user_builds/pvlib-python/checkouts/stable/pvlib/clearsky.py in detect_clearsky(measured, clearsky, times, infer_limits, window_length, mean_diff, max_diff, lower_line_length, upper_line_length, var_diff, slope_dev, max_iterations, return_components)
    823     if len(times) < samples_per_window:
    824         raise ValueError(f"times has only {len(times)} entries, but it must \
--> 825                            have at least {samples_per_window} entries")
    826 
    827     # generate matrix of integers for creating windows with indexing

ValueError: times has only 30 entries, but it must                            have at least 50 entries
```

Presumably the problem is that this code is passing the value `10` by position to the new `infer_limits` parameter added in #1784 instead of to `window_length` like it used to.  I expect it would be fixed by editing that line in [clearsky.rst](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/user_guide/clearsky.rst?plain=1#L645) to specify the parameter by name: `window_length=10`.

matsuobasho, 2023-11-20:
I'd like to tackle this issue.
Seems like your fix suggestion worked (output after running `make html`:

[image]

I have a question about PR's since this is only my second issue.  What is the procedure for updating my fork of the repo as I move from issue to issue?  Should I delete each such branch and create a new fork for a new issue (a little inefficient).  Right now my branch is actually ahead of main by 10 commits, I think because a PR gets squashed into 1 commit. 

Thanks.

cwhanse, 2023-11-20:
>Should I delete each such forked repo and create a new fork for every new issue

I update my main branch, then create a new branch for each PR. I don't think deleting your fork and creating a new one is needed.

If you don't make new branches, then the new PR may contain commits in other PRs that haven't yet been merged.

matsuobasho, 2023-11-21:
@cwhanse , that makes sense, thanks.  Should I also commit the updated clearsky.html file as part of the PR?

PR #1915 (2023-11-21): Fix error in clear-sky user's guide docs page
 - [X] Closes #1914
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [X] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [X] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`nu [... cut]
closed 2023-11-21 by kandersolar