pvlib/pvlib-python #1926: Replacing pandas with polars

langestefan, 2023-12-03:
**Is your feature request related to a problem? Please describe.**
Pandas is used extensively in pvlib all over the place. However after using it for many years I think it is now outdated and needs to be replaced. It is greedy, slow, inconsistent and the API is awkward and difficult to use. 

**Describe the solution you'd like**
I would like to propose replacing pandas with [polars](https://github.com/pola-rs/polars). Polars is a DataFrame interface built in Rust with many, many performance benefits. Too many to list them here, so I suggest visiting their page to learn more. 

I know this is a very big change, but I believe it will lead to some serious performance and usability benefits. So there is no real plan yet, I would like to just start the discussion and see what others think. I think a good starting point would be to benchmark some performance-critical code in pandas and polars so we can quantify what the performance benefits would be.

mikofski, 2023-12-03:
>It is greedy, slow, inconsistent and the API is awkward and difficult to use.

I ❤️ this! 🤣

jules-ch, 2023-12-05:
Like @kevinsa5

I'd say it's way to early to support polars instead of pandas. The API is still unstable, version 0.X.
I'm a big user of polars on personal projects because of performance lazyloading, async collection, integration with the Apache Arrow Ecosystem.

But I really don't see any improvement in using polars here, apart from using the new shiny library, an effort is being made to use numpy array as inputs in functions where pandas were used before see #1455.
Performance-wise, benchmarks need to be made first and since we're not processing GB of data, we probably won't see any difference in performance, especially since we can use the new interface with the Arrow ecosystem with pandas 2.0.

Even when processing 20 years of weather data for a single location it's not even close to several millions of rows.
IMO I don't think it will make a significant gain in perf but I'm eager to see some benchmarks.

Moving forward using Numpy types instead of Dataframe or Series will make integrating with Polars or any Dataframe library easy.

We can probably use polars already if that's what you want with a bunch of PVLIB API.

closed 2024-05-24 by AdamRJensen

markcampanelli, 2024-12-21:
I think it would be more appropriate to simply make pvlib "agnostic" to if one is using `pandas` vs. `polars` (vs. just plain old `numpy` float vectors and a Python `list` of timestamps). Not sure if this could be done with some "timestamped data provider" interface abstraction to lighten the load on users. Given the current evolution of pvlib, this would most likely be painful and disruptive to the `pandas` lovers. Also, sorting all this out would be quite the lift, I fear.

langestefan, 2024-12-21:
> I think it would be more appropriate to simply make pvlib "agnostic" to if one is using `pandas` vs. `polars` (vs. just plain old `numpy` float vectors and a Python `list` of timestamps).

[Narwhals ](https://github.com/narwhals-dev/narwhals) would be way to go.

markcampanelli, 2024-12-21:
> > I think it would be more appropriate to simply make pvlib "agnostic" to if one is using `pandas` vs. `polars` (vs. just plain old `numpy` float vectors and a Python `list` of timestamps).
> 
> [Narwhals ](https://github.com/narwhals-dev/narwhals) would be way to go.

Oh cool. I need to check this out further!