pvlib/pvlib-python #1928: v0.10.3 release plan

cwhanse, 2023-12-08:
We expect to issue v0.10.3 before the end of 2023.

PRs tagged with v0.10.3:

- [ ] #1924
- [ ] #1907 
- [ ] #1905 
- [ ] #1904 
- [ ] #1893 
- [ ] #1886 
- [ ] #1883 
- [ ] #1828 
- [ ] #1827 
- [ ] #1725 

Issues tagged wtih v0.10.3

- [ ] #1922 
- [ ] #1901 will close with #1907
- [ ] #1870 
- [ ] #1856 it looks like #1884 won't be ready for v0.10.3
- [ ] #1824 will close with #1827
- [ ] #1734 it doesn't look like #1725 will close this, maybe defer @mikofski 
- [ ] #1690 would close with #1725
- [ ] #1689 no progress since 03/2023

echedey-ls, 2023-12-11:
I won't be able to work on #1904 until February; I haven't been around here since college has been and will be tough until then.
Related issues to it (#1734, #1725) are on my personal roadmap; again, when I get back to this.

AdamRJensen, 2023-12-12:
I had a brief discussion with @kandersolar and made a list of suggested PRs to be deferred (see update to Cliff's list) above.

PR #1934 (2023-12-20): Finalize 0.10.3
 - [x] Closes #1928
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or th [... cut]
closed 2023-12-20 by kandersolar