pvlib/pvlib-python #1929: Usage of more efficient Sandia Modules?

k-rieck, 2023-12-13:
Dear PVLib Team,

we would like to use more efficient PV modules with efficiencies of 21-22% as they are represented in your CEC module database (e.g. "Trina Solar TSM-410DE09")  
Your pvlib functions (e.g pvlib.pvsystem.samp) seem to work only with Sandia modules which seem to have lower efficiencies.
Is there a way to use the pvlib functions on more efficient modules?

Many thanks in advance!

kandersolar, 2023-12-13:
Yes, these modules can be simulated with pvlib.  To use parameters taken from the CEC module database, you must use the CEC performance model, which is available in pvlib with `pvlib.pvsystem.calcparams_cec` and `pvlib.pvsystem.singlediode`.  `ModelChain` can use CEC module parameters as well.

- https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.calcparams_cec.html
- https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.singlediode.html
- https://pvlib-python.readthedocs.io/en/stable/user_guide/pvsystem.html#module-and-inverter-parameters

k-rieck, 2023-12-13:
Thanks a lot! We will check it out

k-rieck, 2023-12-13:
Hi, I tried to follow the example in the link https://pvlib-python.readthedocs.io/en/stable/gallery/iv-modeling/plot_singlediode.html#sphx-glr-gallery-iv-modeling-plot-singlediode-py.
Instead of using "pvsystem.calcparams_desoto" I tried "pvlib.pvsystem.calcparams_cec".
Is there a way to calculate the effective irradiance like in "pvlib.pvsystem.sapm_effective_irradiance" for the cec modules?
Thanks a lot!

cwhanse, 2023-12-13:
Effective irradiance is POA global irradiance reduced by reflections of the direct component and adjusted for spectrum.

You can use the SAPM bundled approach, but that requires coefficients you probably don't have.

You can use functions in `pvlib.iam` to get the reflection loss multiplier, and functions in `pvlib.spectrum` to get the spectral adjustments.

This process could really use a good example in the Gallery, and maybe, a convenience function in `pvlib.pvsystem.`