pvlib/pvlib-python #1942: Detailed explanation of `racking_module` and `module_type` in `PVSystem` docstring

matsuobasho, 2023-12-26:
**Is your feature request related to a problem? Please describe.**
Without prior knowledge, the `racking_model` and the `module_type` parameters are not properly explained, especially with respect to how their combination is used to select a temperature model.  

**Describe the solution you'd like**
A detailed description `racking_model` and `module_type` and and how/why certain combinations then are used to select a SAPM or PVSyst temperature model.  This description should be in the notes section of `PVSystem` and `Array`.  

It would also be useful to include a brief version of this description in the `PVSystem` [user guide](https://pvlib-python.readthedocs.io/en/stable/user_guide/pvsystem.html).  

Also, note that currently the docstring of the `racking_model` argument in both `PVSystem` and `FixedMount` is incomplete - it is  missing the `freestanding` option.  This needs to be fixed.

**Describe alternatives you've considered**
I've searched the codebase and tutorials for an explanation.

matsuobasho, 2024-02-06:
In addition to the `racking_model` incompleteness mentioned above, the docstring [states ](https://github.com/pvlib/pvlib-python/blob/main/pvlib/pvsystem.py#L172) that default is 'open_rack', but the [code ](https://github.com/pvlib/pvlib-python/blob/main/pvlib/pvsystem.py#L204) has the default as `None`.  Which is one the intended definition?

cwhanse, 2024-02-15:
IMO `None` is the intended default and the docstring should be corrected.