pvlib/pvlib-python #1950: Add new spectral factor models

RDaxini, 2024-01-19:
****Is your feature request related to a problem? Please describe.****
pvlib currently includes three spectral factor models: First Solar, Caballero, Sandia. It could be helpful to expand this range to include models based on different indices and/or parameterisations.

****Describe the solution you'd like****
I would start by working on implementing the PVSPEC model [1], which has demonstrated good performance relative to the First Solar approach that is already included in pvlib. PVSPEC is based on air mass and clearness index, the latter of which is not used in the three existing spectral factor models in pvlib.

I could develop this later to include different parameterisations of the air mass/clearness index approach (e.g. [2]) and new models based on different parameters.

****Additional context****
1) Are there any particular models that you would recommend prioritising? 
2) I am working off the models in this review [3], are there any models not covered there that would be beneficial to include in pvlib?

Refs:
[1] https://doi.org/10.1109/PVSC45281.2020.9300932
[2] https://www.researchgate.net/publication/256080247_A_simple_model_for_estimate_the_influence_of_spectrum_variations_on_PV_performance
[3] https://doi.org/10.1016/j.energy.2023.129461

cwhanse, 2024-01-19:
@RDaxini thanks for opening this issue. Nice paper.

1. The PVSEC model would be a nice addition to pvlib. As to other models surveyed in your paper, I would prioritize those which don't rely on spectral irradiance measurements, although the more detailed models are also of (lesser) interest. Older, simpler models are primarily of historical interest.
2. As to other models, given the analysis in your paper, I would defer to your recommendations, provided that 1) there is a published reference fully describing the candidate model and 2) the publication includes some validation of the proposed model.

echedey-ls, 2024-03-06:
Hi @RDaxini !

Have you done any progress on this? I would like to contribute this issue (or part of it) for this year GSoC program.

AdamRJensen, 2024-03-06:
@echedey-ls I know that @RDaxini already has a GSoC application planning on this topic for this year.

echedey-ls, 2024-03-06:
@AdamRJensen oh nice to know that!

RDaxini, 2024-07-28:
@AdamRJensen @kandersolar I think this part of my GSoC is complete now since the APE_e model is not being implemented at this stage. Should we close the issue?

closed 2024-07-28 by AdamRJensen