pvlib/pvlib-python #1952: Gallery examples of ModelChains for the DOE Solar Data Bounty Prize systems

kandersolar, 2024-01-24:
The U.S. Department of Energy sponsored a [Solar Data Bounty Prize](https://www.energy.gov/eere/solar/american-made-solar-data-bounty-prize) that resulted in publication of detailed monitoring data and system configuration specs for five commercial PV systems.  I think we should create a section in the example gallery containing "reference models" for these systems.  This section could be expanded over time to have models corresponding to other public PV datasets too (RTC, NIST, etc).

System descriptions:
- https://openei.org/wiki/PVDAQ/Sites/Farm_Solar_Array
- https://openei.org/wiki/PVDAQ/Sites/Maui_Ocean_Center
- https://openei.org/wiki/PVDAQ/Sites/SR_CO
- https://openei.org/wiki/PVDAQ/Sites/Shine_On_Solar_Facility
- https://openei.org/wiki/PVDAQ/Sites/Simon_Solar_Farm

Data available for download here: https://data.openei.org/s3_viewer?bucket=oedi-data-lake&prefix=pvdaq%2F2023-solar-data-prize%2F

It would be great to get these pages created and merged before March 21 so that they can be mentioned in the [Modeling Tools webinar](https://www.energy.gov/eere/solar/solar-energy-technologies-office-events-and-webinars), so I'm tagging this for v0.10.4 (scheduled for mid-March).

AdamRJensen, 2024-02-06:
Did DOE put funds aside for such work (seems like that would've made sense)?

kandersolar, 2024-02-06:
To my knowledge, no special funding is currently available for this work.

williamhobbs, 2024-02-20:
I _really_ like this idea.

SR_CO looks like thin-film with true-tracking, and I think Simon Solar Farm is fixed tilt 2P crystalline, so both could be good for validation of self-shade models.

kandersolar, 2024-02-21:
A note for whoever makes these models: it's a little hard to tell, but from google street view it looks like SR_CO is a 4L array, so don't be surprised if the simple linear self-shading model doesn't perform very well. 

[image] 

[image]

https://www.google.com/maps/@40.3848262,-104.547199,3a,15y,351.87h,88.67t/data=!3m6!1e1!3m4!1szpxv4gM0fGnA5kIHnMoA1Q!2e0!7i16384!8i8192?entry=ttu

kandersolar, 2024-03-06:
Here is a rough example model page for one system (SR_CO): https://gist.github.com/kandersolar/2ab973ea0b5d8cd45531a38c6e1fd06f

Feedback is welcome.

One general consideration for including this kind of example page in the docs is that to actually run the model and perform the comparison requires a lot of i/o (fetching PSM3 data and fetching the measured power data).  So far we have managed to avoid performing significant i/o in the example gallery, and it would be great to keep it that way.  I suggest we configure these particular examples to not be executed during the docs build, and just include the comparison plots as static PNGs.  We could also just omit the plots entirely, but IMHO the figures make the example page a lot more engaging.

adriesse, 2024-03-06:
Looks good to me. I'm surprised that you  have to guess at metadata given that that these are presumably the five winners.  Are they?

If we settle on a way to deal with large data in examples, then we could migrate some examples from DuraMAT to pvlib as well.

williamhobbs, 2024-03-06:
Regarding guessing metadata, I think that a lot of array geometry info (pitch, collector width, GCR, tracker height) was left off of the list of requested metadata for the bounty prize. And manufacturer/model info was optional to provide. 

@kandersolar: all of your assumptions look reasonable to me.

mikofski, 2024-03-07:
Why not move the gallery to it’s own repo?

kandersolar, 2024-03-07:
> Why not move the gallery to it’s own repo?

I guess these models could be a repository alongside the other [tutorial repositories](https://pvsc-python-tutorials.github.io/), but (1) I don't envision these to really be tutorials, just examples, and (2) I don't think that would completely solve the i/o problem anyway -- the Data Prize datasets are pretty large (some 100s of MB), which is a lot to be downloading or storing no matter what repository the code lives in. 

> given that that these are presumably the five winners. Are they?

Yep, those are the winners.  It's too bad that more detailed design information wasn't collected, but at least it's possible to cobble together estimates for enough missing pieces to make a simple but still reasonable model.

adriesse, 2024-03-07:
Perhaps we don't really need those examples for pvlib purposes, but in your opener you also called them "reference models".  I think it would be a great open source/open data/open science achievement to run with the partial published data; scrape, glean and reverse engineer the missing bits and meta-bits; and then publish a complete reference model with documentation for each!