pvlib/pvlib-python #1955: AOI dependence for soiling losses

kandersolar, 2024-01-29:
**Is your feature request related to a problem? Please describe.**
Soiling losses are known to show a dependence on angle of incidence, but pvlib's soiling loss models don't consider AOI.

**Describe the solution you'd like**
Implement a model that calculates an adjustment factor, analogous to IAM, that converts soiling loss at AOI=0 to soiling loss at any angle of incidence.

Here is one such model: Guo & Javed 2024, Effect of incidence angle on PV soiling loss.  https://doi.org/10.1016/j.solener.2023.112298

**Describe alternatives you've considered**
N/A

**Additional context**
One wrinkle: soiling loss models tuned to field measurements probably don't calculate soiling loss at AOI=0, but rather some kind of "average" AOI that depends on both the sun position and the weather conditions (sun AOI matters less in cloudy climates).  Conceptually, it seems like soiling loss ought to be considered a combination of direct and diffuse components the same way IAM is.  It's not obvious to me how best to apply an AOI correction factor to empirical soiling models.

echedey-ls, 2024-03-02:
I've read the paper and I don't dislike the proposed model. The only issues I have with it are:
1. Do we want to provide some kind of fitting functions so the two parameters can be obtained from some kind of curve fit? The paper approach is for some exact soiling values but in the field they change with the weather.
   - In this case, what data can/should I make use of?
   - The $SL_0$ can be left up to the user but I can't think of how to get it from this kind of data.
     I have in mind this data could be two flat reference cells, one that cleans itself (or is cleaned) everyday and another one that gets soiled normally. So at AOI=0º, $SL_0$ is direct. For other cases, I'm not sure.
    Rearranging (7) as $SL_\theta(\theta) = SL_0 · (sec \theta)^P [\frac{1 - F_n/IAM}{1 - F_n}]$ there is another parameter, $SL_0$. Whether this works or not, I'm 0% sure.
2. Data testing - I can do the average functional test of datatypes and some numbers, but they won't guarantee scientific rigour. Maybe from the graphs I can approximate some values...

mikofski, 2024-03-02:
ICYMI: @wholmgren , @abhisheksparikh , @mdeceglie

adriesse, 2024-03-04:
I don't have access to this paper, but I do recall that Martin & Ruiz discussed representing the dirty modules with a different a_r.  That wouldn't accommodate losses at normal incidence.

CameronGordonn, 2026-11-12:
Following up on the Table 2 question: the values come from [18], the DTU dataset (Guo's text cites [18,19], the table header only [19]). They're the mean of Lab A's four samples and match to all 7 decimals; the 90° value of 0 is Guo's addition:

import pandas as pd
df = pd.read_csv('AOIRR2020RawData.csv', encoding='utf-8-sig')
df.columns = ['lab', 'sample', 'type', 'aoi', 'tau']
lab_a = df[df['lab'] == 'A'].groupby('aoi')['tau'].mean()
print(lab_a.loc[[0, 45, 60, 70, 75, 80, 85]].round(7).tolist())

Since Fn and P were fitted with this table, I'd keep it as the default, citing the dataset as its terms ask. Compared with iam.physical in @williamhobbs's sketch, the modeled loss differs by up to 2% through 65° but by about 16% at 75°.

On P: the Fig. 5 fit for this study (P = 31.48·SL0² − 9.88·SL0 + 1.57) bottoms out at P ≈ 0.80 near SL0 = 0.157 and reaches 1.57 at SL0 = 0, so it shouldn't be extrapolated. Interpolating Table 3 with np.interp holds P at 1.15 below 0.05 and at 0.80 above 0.15, consistent with the 0.8 cap suggested above. Alternatively, the paper reports that P = 1 (or 0.9) with Fn = 0.7 gives daily loss within 6% (Table 5).

If nobody else is planning to, I'd be glad to put together a PR once the defaults are settled.