pvlib/pvlib-python #1958: datetime64[ns] and datetime64[s]

peterdudfield, 2024-01-31:
I found a different when using `datetime[ns]` and `datetime[s]`

pvlib version == 0.10.2
numpy version == 1.23.2

```
import pvlib
import numpy as np

dt = np.array([datetime(2024,1,31, 12,0), datetime(2024,1,31,12,30)], dtype='datetime64[s]')

solpos = pvlib.solarposition.get_solarposition(
            time=dt,
            latitude=50,
            longitude=0)
```

Perhaps I don't fully understand the different between [ns] and [s] but would be great if anyone can help on this

https://github.com/openclimatefix/ocf_datapipes/pull/272

cwhanse, 2024-01-31:
Thanks @peterdudfield this results from pandas > 2.0 #1932 and is being accomodated in pvlib with #1948

peterdudfield, 2024-01-31:
Thanks @cwhanse for the quick response

kandersolar, 2024-02-23:
Duplicate of #1932, fixed by #1948

closed 2024-02-23 by kandersolar