pvlib/pvlib-python #1965: CECMod modules return ValueError when run ModelChain

itsklimov, 2024-02-08:
**Describe the bug**
if I use cells from pvlib.pvsystem.retrieve_sam("CECMod")
I get:
ValueError: could not infer AOI model from system.arrays[i].module_parameters. Check that the module_parameters for all Arrays in system.arrays contain parameters for the physical, aoi, ashrae, martin_ruiz or interp model; explicitly set the model with the aoi_model kwarg; or set aoi_model="no_loss".

**To Reproduce**
Steps to reproduce the behavior:
setup:
```
from pvlib.location import Location
from pvlib.pvsystem import PVSystem
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS
import pvlib
from pvlib.modelchain import ModelChain

pv_location = Location(
    latitude=PV_LATITUDE, longitude=PV_LONGITUDE, tz=TZ, altitude=PV_ALTITUDE
)
temperature_model_parameters = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']
cec_modules = pvlib.pvsystem.retrieve_sam("CECMod")
cec_module = cec_modules[
    "Mission_Solar_Energy_LLC__MSE380SQ7S"
]
cec_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')
cec_inverter = cec_inverters["Yaskawa_Solectria_Solar__PVI_50TL_480__480V_"]

system = PVSystem(
    name="H3Mini",
    surface_tilt=PV_SURFACE_TILT,
    surface_azimuth=PV_SURFACE_AZIMUTH,
    # module=sandia_module,
    module_parameters=cec_module,
    # inverter=cec_inverter,
    inverter_parameters=cec_inverter,
    modules_per_string=12,
    strings_per_inverter=2,
    module_type="glass_glass",
    temperature_model_parameters=temperature_model_parameters,
)

mc = ModelChain(system, pv_location)
```

**Expected behavior**
expected to have the model to perform calculations

**Versions:**
 - ``pvlib.__version__``: 0.10.3
 - ``pandas.__version__``: 2.0.3
 - python: 3.11

How to fix it?

cwhanse, 2024-02-08:
@itsklimov the CEC module database doesn't contain parameters for any of the AOI models. You'll need to add those parameters to `module_parameters` for the AOI model you choose.

Either
```
mc = ModelChain(system, pv_location, aoi_model='no_loss')
```

which skips the AOI model, or set `aoi_model` to be one of 'physical', 'ashrae', 'sapm', 'martin_ruiz', 'interp' and add the parameters for that model's function in `pvlib.iam`.

@pvlib/pvlib-core Since several AOI models have default parameters, we could add a bit more intelligence here and use default parameters when such a model is specified but no parameters are given.

itsklimov, 2024-02-08:
I solved it based on https://github.com/pvlib/pvlib-python/issues/621
I added relative_humidity to the weather forecast and used
weather["precipitable_water"] = pvlib.atmosphere.gueymard94_pw(weather["temp_air"], weather["relative_humidity"])  # needed for aoi_model="physical"

itsklimov, 2024-02-08:
sorry for the dumb question, but how to determine which of the ‘physical’, ‘ashrae’, ‘martin_ruiz’ models best fit for my setup?

cwhanse, 2024-02-08:
'physical' and 'martin_ruiz' are more popular, the results are quite close for these two models. Without measurements of the reflections from that Mission Solar module I don't think there's much basis to prefer one model over another.

cwhanse, 2024-02-08:
> I solved it based on #621 I added relative_humidity to the weather forecast and used weather["precipitable_water"] = pvlib.atmosphere.gueymard94_pw(weather["temp_air"], weather["relative_humidity"]) # needed for aoi_model="physical"

Just to clarify, `preciptible_water` is used for the spectral mismatch model, not the AOI model.

cwhanse, 2024-02-08:
> > Since several AOI models have default parameters, we could add a bit more intelligence here and use default parameters when such a model is specified but no parameters are given.
> 
> If I understand the idea correctly, I think this already works. See this example where `aoi_model='physical'` is specified but no parameters are supplied: https://pvlib-python.readthedocs.io/en/stable/gallery/irradiance-transposition/plot_mixed_orientation.html

So it does. Specify `aoi_model` and the code skips the inference step.

closed 2024-02-08 by cwhanse