pvlib/pvlib-python #1966: Uncertainty about using timezone locale vs UTC in pandas timestamps for solar position

markcampanelli, 2024-02-09:
Using `pvlib-python` v0.10.3, the docstring for the `time` parameter for [get_solarposition()](https://github.com/pvlib/pvlib-python/blob/main/pvlib/solarposition.py#L44) is

>     time : pandas.DatetimeIndex
>         Must be localized or UTC will be assumed.

I interpreted this that one should be able to provide a `pandas.DatetimeIndex` with a well-defined timezone and the spring and fall offsets for daylight savings would be properly handled by the solar position algorithms. However, I later found that these offsets caused errors in the solar position calculation. When I used UTC from the start (which avoids daylight savings offsets), the errors went away.

I realize that this is likely a common user error, but I did think I was following the correct way of using timezone-aware timestamps.

Am I correct that daylight savings offsets should be able to be handled by the solar position algorithms?

Perhaps I did not properly assign the locale in `pandas.DatetimeIndex` I created?

kandersolar, 2024-02-09:
The effect of DST is included if you use a time zone that includes DST transitions.  `US/Mountain` includes DST while `Etc/GMT+7` does not.  See https://pvlib-python.readthedocs.io/en/stable/user_guide/timetimezones.html#daylight-savings-time

Here's an example showing how different time zones do and don't display DST dislocations:

[image]

```python
import pandas as pd
import matplotlib.pyplot as plt
import pvlib

def show_dstplot(signal, ax):
    temp = pd.DataFrame({
        'values': signal,
        'time of day': signal.index.hour + signal.index.minute / 60,
        'day of year': signal.index.dayofyear,
    })
    pivot = temp.pivot_table(columns='day of year', index='time of day', values='values')
    ax.pcolormesh(pivot.columns, pivot.index, pivot.values)

tz_trials = [None, 'UTC', 'Etc/GMT+5', 'US/Eastern']

fig, axes = plt.subplots(len(tz_trials), 1, sharex=True,figsize=(4, 8))

for tz, ax in zip(tz_trials, axes):
    times = pd.date_range('2019-01-01 00:00', '2019-12-31 23:59', freq='5min', tz=tz)
    sp = pvlib.solarposition.get_solarposition(times, 40, -80)
    show_dstplot(sp['elevation'].clip(lower=0), ax)
    ax.set_title(f'{tz=}')
    ax.set_ylabel('Hour of day')

axes[-1].set_xlabel('Day of year')
fig.tight_layout()
```

markcampanelli, 2024-02-11:
@kandersolar Thanks for the helpful explanation. Using `US/Mountain` instead of UTC with the +7 offset was indeed my issue.

TBH, I think the `pvlib-python` instruction "Use UTC." would be super easy to follow and much harder to screw up.

Closing this.

closed 2024-02-11 by markcampanelli