pvlib/pvlib-python #1975: retire support for python 3.8

cwhanse, 2024-02-13:
Python 3.7 reached end-of-life on 2023-06-27. I propose we remove support for python 3.7 with pvlib-python v0.11.0

AdamRJensen, 2024-02-13:
Sounds good to me.

adriesse, 2024-02-13:
What's the rush?

kandersolar, 2024-02-14:
One reason to drop 3.7 is to prevent excessive accumulation of supported versions: python releases new versions annually, and it's been more than a year since we dropped a version.

alexandermorgan, 2024-03-14:
If you make 3.8 your minimum supported version you could also use some of the features that came in 3.8. This [_numdiff](https://github.com/pvlib/pvlib-python/blob/92c0e5bcfca717056971c27c556ff443d07d4e4c/pvlib/ivtools/utils.py#L18) function does a bunch of math with lots of repeated chunks. For example, `a0[:, 0] * a0[:, 1]` is calculated 9 times. You could set up a mini library of chunks and then reference those variables now, but a more concise way would py3.8's walrus operator. The first reference of the above gets simultaneously assigned to a variable:
`(col0Xcol1 := a0[:, 0] * a0[:, 1])`
Then all the subsequent repetitions use that `col0Xcol1` variable. There would be other things too, like being able to reverse a dictionary. I've seen other libraries that have an official policy of supporting the last X python versions. X = 6 (i.e. py 3.7 - 3.12) seems like a lot.

kandersolar, 2024-04-30:
I suggest we move this item up to 0.10.5, and plan to release 0.10.5 soon (perhaps even this week) for #2026.  

Waiting for 0.11.0 would put the desire to drop 3.7 ASAP (#2023) in conflict with wanting to wait at least a year before removing the 0.10.0 (released June 2023) deprecations set to expire in 0.11.0.

PR #2029 (2024-05-03): Increase python requirement to >= 3.8
 - [x] Closes #1975, closes #2023
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitH [... cut]
closed 2024-05-03 by kandersolar