pvlib/pvlib-python #1976: Ambiguous descriptions of axis_azimuth and axis_tilt in pvlib.tracking.singleaxis() docs

williamhobbs, 2024-02-20:
**Is your feature request related to a problem? Please describe.**
Some details around _axis_azimuth_ and _axis_tilt_ in the docs (https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.tracking.singleaxis.html) seem ambiguous to me. 

Examples:

1. For a north-south aligned tracker axis, sloping 5 degrees down to the south, what are the inputs for _axis_azimuth_ and _axis_tilt_? I assume _axis_azimuth_ = 180, _axis_tilt_ = 5.
2. For the same north-south aligned tracker axis, but sloping 5 degrees down to the **_north_**, what are the inputs? I assume _axis_azimuth_ = 0, _axis_tilt_ = 5, since _axis_tilt_ must be >= 0.

Additionally, the description in the docs includes:
> For example, if tracker _axis_azimuth_ is 180 (oriented south) and _axis_tilt_ is zero...

But the default for _axis_azimuth_ is 0, which could lead to confusion around inputs and output sign convention. 

**Describe the solution you'd like**
More clear docs.

**Additional context**
Changing definitions slightly and allowing for negative _axis_tilt_ would make more sense to me, but maybe that causes other problems. 

For example \# 2 above (sloping down to the North), allowing for _axis_azimuth_ = 180, _axis_tilt_ = -5 makes more since to me. You could unambigously define _axis_azimuth_ as "the South-most direction of the axis of rotation" and _axis_tilt_ as "the tilt of the axis of rotation, with positive values indicating downward slope towards the South-most direction, and negative values indicating downward slope towards the North-most direction...".

I'm not sure how to handle the odd case of a tracker axis aligned perfectly East-West.

And for systems in the Southern Hemisphere, I don't think anything would need to change.

williamhobbs, 2024-02-20:
@mikofski, I agree. Good point. 

Keeping a consistent rotation convention makes more sense to me than (the way I interpret) the existing docs for `pvlib.tracking.singleaxis()`: _tracker_theta_ sign reverses depending on whether you pick 0 or 180 for _axis_azimuth_.

janinefreeman, 2024-02-20:
I left a comment on the SAM issue as well, but allowing negative input tilts would require some additional mathematical gymnastics inside the functions to make angle calculations and shading calculations work out (I'm assuming this is true for pvlib as well) - at the very least, transposition to the expected case before doing the math (i.e. if a user enters azimuth = 180, tilt = -5, the code would need to translate that to azimuth = 0, tilt = 5, then correctly re-translate the outputs). I have wondered about this issue of "how to explain which end of the tracker is the rotation around" before, since right now it follows mathematical rotational axis definitions, rather than cardinal direction conventions, that certainly aren't intuitive for most. Open to ideas here on the best way to communicate what tracker rotation angles mean...

williamhobbs, 2024-02-20:
Keeping conventions the same between pvlib and SAM seems like it would be very beneficial to users. 

The rotation "convention" of negative in the morning, positive in the afternoon, would stay reversed in the case of sloping down to the North, but 1) this can be clarified in docs, and 2) is likely a pretty rare case. I think anyone that is modeling an atypical system like this and needs to carefully interpret _tracker_theta_ is already pretty far into the "weeds" and can probable make sense of it all with decent docs.

mikofski, 2024-02-20:
@williamhobbs  please just try it with negative `axis_tilt` and see if it works already as expected. I looked through the code here: https://github.com/pvlib/pvlib-python/blob/main/pvlib/tracking.py and I don’t see anything other than the docstring that limits it between zero and 90°. @janinefreeman I believe pvlib should just work even if axis tilt is negative without any changes at all because `singleaxis()` used `arctan2` instead of cosine so should work in any quadrant.

AdamRJensen, 2024-02-20:
I would vote in favor of allowing negative tilts, and thus allowing the standard convention of negative morning and positive evening tilts.

I could easily imagine simulations of larger systems (where the field is broken down into several sub-systems of uniform tilt) where one part of the field is sloping north and another is sloping south. In such a case it would be inconvenient having different axis azimuth conventions for different parts of the field.

williamhobbs, 2024-02-20:
@mikofski you are correct, it works with negative _axis_tilt_.

[image]

williamhobbs, 2024-02-20:
Since negative values of _axis_tilt_ are allowed in the code, it seems like users can use either convention and the docs simply need to be updated accordingly. And part of that is noting that if users select _axis_azimuth_ between 270 and 90 (North-facing), the rotation angle convention is reversed.

cwhanse, 2024-02-20:
I'm OK with negative axis tilt, as long as the definition of axis_tilt and axis_azimuth are not dependent on hemisphere, and together unambiguously define the coordinate system that defines rotation. I'd be opposed to starting with a defined rotation scheme (e.g. negative is east) because I'm fairly sure that will make the coordinate system depend on hemisphere.

matsuobasho, 2024-03-19:
Taking the above discussion into account, if we specify an `axis_azimuth` of 0 and an `axis_tilt` of -5, what does that mean in terms of the positioning of the trackers?

From the way the docstring is currently worded
`The tilt of the axis of rotation (i.e, the y-axis defined by axis_azimuth) with respect to horizontal.`
seems like it would be a confusing situation where we specify that the azimuth is 0 (point towards north), but also specify that there is a -5 degree tilt, meaning that the slope is actually towards the south?

markcampanelli, 2026-02-12:
I set up what I think is a typical computational workflow using the ground slope, slope azimuth, and axis azimuth to first compute the axis tilt and cross-axis tilt using `pvlib.tracking.calc_axis_tilt` and `pvlib.tracking.calc_cross_axis_tilt`, respectively. Then, these values were fed into `pvlib.tracking.singleaxis`. I tried to set up the same geometry (here, a northern hemisphere setup with the array axis sloping away from the equator), using the two 180-degree opposing choices for axis azimuth. I see the expected change in sign for the rotation angles, but the computations from `pvlib.tracking.singleaxis` appear to be different (esp. different AOIs). I think they should have been the same, but I may be missing something.

First representation:
```python
slope_azimuth = 0
print(f"Slope azimuth: {slope_azimuth:0.01f}°")
slope_tilt = 10
print(f"Slope tilt: {slope_tilt:0.01f}°")
axis_azimuth = 180
print(f"Axis azimuth: {axis_azimuth:0.01f}°")

# calculate the tracker axis tilt, assuming that the axis follows the terrain:
axis_tilt = pvlib.tracking.calc_axis_tilt(slope_azimuth, slope_tilt, axis_azimuth)
print(f"Axis tilt: {axis_tilt:0.01f}°")

# calculate the cross-axis tilt:
cross_axis_tilt = pvlib.tracking.calc_cross_axis_tilt(
    slope_azimuth, slope_tilt, axis_azimuth, axis_tilt
)
print(f"Cross-Axis tilt: {cross_axis_tilt:0.01f}°")

apparent_zenith = numpy.array((45, 30, 45))
apparent_azimuth = numpy.array((135, 180, 225))

pvlib.tracking.singleaxis(
    apparent_zenith,
    apparent_azimuth,
    axis_tilt=axis_tilt,
    axis_azimuth=axis_azimuth,
    max_angle=90,
    backtrack=False,
    gcr=2.0 / 7.0,
    cross_axis_tilt=cross_axis_tilt,
)
```

Outputs:
pvlib.tracking.calc_axis_tilt: -10.0°

pvlib.tracking.calc_cross_axis_tilt: -0.0°

pvlib.tracking.singleaxis:
```
{'tracker_theta': array([-39.36170387,   0.        ,  39.36170387]),
 'aoi': array([22.68560889, 20.        , 22.68560889]),
 'surface_azimuth': array([101.9525464, 180.       , 258.0474536]),
 'surface_tilt': array([40.41119575, 10.        , 40.41119575])}
```

Second representation:
```python
slope_azimuth = 0
print(f"Slope azimuth: {slope_azimuth:0.01f}°")
slope_tilt = 10
print(f"Slope tilt: {slope_tilt:0.01f}°")
slope_azimuth = 0
slope_tilt = 10
axis_azimuth = 0
print(f"Axis azimuth: {axis_azimuth:0.01f}°")

# calculate the tracker axis tilt, assuming that the axis follows the terrain:
axis_tilt = pvlib.tracking.calc_axis_tilt(slope_azimuth, slope_tilt, axis_azimuth)
print(f"Axis tilt: {axis_tilt:0.01f}°")

# calculate the cross-axis tilt:
cross_axis_tilt = pvlib.tracking.calc_cross_axis_tilt(
    slope_azimuth, slope_tilt, axis_azimuth, axis_tilt
)
print(f"Cross-Axis tilt: {cross_axis_tilt:0.01f}°")

apparent_zenith = numpy.array((45, 30, 45))
apparent_azimuth = numpy.array((135, 180, 225))

pvlib.tracking.singleaxis(
    apparent_zenith,
    apparent_azimuth,
    axis_tilt=axis_tilt,
    axis_azimuth=axis_azimuth,
    max_angle=90,
    backtrack=False,
    gcr=2.0 / 7.0,
    cross_axis_tilt=cross_axis_tilt,
)
```

Outputs:
pvlib.tracking.calc_axis_tilt : 10.0°

pvlib.tracking.calc_cross_axis_tilt: 0.0°

```
pvlib.tracking.singleaxis:
{'tracker_theta': array([ 3.93617039e+01,  4.57983173e-15, -3.93617039e+01]),
 'aoi': array([37.96584945, 40.        , 37.96584945]),
 'surface_azimuth': array([7.80474536e+01, 2.63741998e-14, 2.81952546e+02]),
 'surface_tilt': array([40.41119575, 10.        , 40.41119575])}
```

williamhobbs, 2026-02-12:
@markcampanelli, I'm seeing similar behavior. When I previously tried "flipping around" the azimuth and tilt, I only looked at tracker theta, not AOI and surface orientation. That doesn't seem expected. 

[image]

[image]

[image]

```python
latitude, longitude = 36.397, -114.962
times = pd.date_range('2025-12-21 13:00', '2025-12-22 2:00', freq='1min', tz='UTC')
sp = pvlib.solarposition.get_solarposition(times, latitude, longitude)

tr1 = pvlib.tracking.singleaxis(
    apparent_zenith=sp.apparent_zenith,
    solar_azimuth=sp.azimuth,
    axis_tilt=0,
    axis_azimuth=180,
    max_angle=52,
)

tr2 = pvlib.tracking.singleaxis(
    apparent_zenith=sp.apparent_zenith,
    solar_azimuth=sp.azimuth,
    axis_tilt=10,
    axis_azimuth=180,
    max_angle=52,
)

tr3 = pvlib.tracking.singleaxis(
    apparent_zenith=sp.apparent_zenith,
    solar_azimuth=sp.azimuth,
    axis_tilt=0,
    axis_azimuth=0,
    max_angle=52,
)

tr4 = pvlib.tracking.singleaxis(
    apparent_zenith=sp.apparent_zenith,
    solar_azimuth=sp.azimuth,
    axis_tilt=-10,
    axis_azimuth=0,
    max_angle=52,
)

tr1.tracker_theta.plot(label='tilt=0, az=180')
tr2.tracker_theta.plot(label='tilt=10, az=180')
tr3.tracker_theta.plot(label='tilt=0, az=0')
tr4.tracker_theta.plot(label='tilt=-10, az=0')
plt.ylabel('tracker_theta')
plt.legend()
plt.show()

tr1.aoi.plot(label='tilt=0, az=180')
tr2.aoi.plot(label='tilt=10, az=180')
tr3.aoi.plot(label='tilt=0, az=0', linestyle='--')
tr4.aoi.plot(label='tilt=-10, az=0')
plt.ylabel('aoi')
plt.legend()
plt.show()

tr1.surface_azimuth.plot(label='tilt=0, az=180')
tr2.surface_azimuth.plot(label='tilt=10, az=180')
tr3.surface_azimuth.plot(label='tilt=0, az=0', linestyle='--')
tr4.surface_azimuth.plot(label='tilt=-10, az=0')
plt.ylabel('surface_azimuth')
plt.legend()
plt.show()
```

cwhanse, 2026-02-13:
AOI looks wrong for tilt=-10 because the calculation of AOI relies on `tracking.calc_surface_orientation` which converts `axis_tilt` to the usual pvlib interpretation of tilt, i.e., a positive angle from horizontal that is ignorant of azimuth. If negative `axis_tilt` is not to be forbidden, this line needs to be corrected: https://github.com/pvlib/pvlib-python/blob/bba49c0c477bdb442bf889f071a7a86d03d1766e/pvlib/tracking.py#L246

markcampanelli, 2026-02-14:
@cwhanse Would you agree that that the addition here of the axis_azimuth ALSO defining a rotation axis is what makes this geometric situation more complex so that "the usual pvlib interpretation of tilt, i.e., a positive angle from horizontal that is ignorant of azimuth" is insufficient?

markcampanelli, 2026-02-19:
I will follow up more generally here to say that I'm likely going to have to address this issue in my own application, so I'm happy to share that work in a PR to the repo. I just want to make sure that the extra complexity is deemed appropriate by the maintainers.

cwhanse, 2026-04-02:
@echedey-ls I'm afraid I don't follow. If you are rotating a rectangle with fixed `surface_azimuth` and decreasing `surface_tilt`, how does `axis_aximuth` enter?

echedey-ls, 2026-04-02:
The hypothetical situation I propose is probably just that, hypothetical. That was the first thing that came to my mind to illustrate that the fact that `surface_azimut=axis_azimuth` is probably unexpected in some cases as it is completely different from the neighboring values `axis_azimuth +/- 90` in the typical HSAT case.

The workflow is the following: `theta` (variable), `axis_azimuth` (fixed) and `axis_tilt` (fixed) are inputs to `pvlib.tracking.calc_surface_orientation`. Surface's azimuth and tilt are obtained, and they are then used to draw this rectangle.

For non-zero `thetas`:
1. -20

[image]

2. -10

[image]

3. 10

[image]

4. 20

[image]

But for `theta=0`:

[image]

<details><summary>Code</summary>
<p>

Search for `theta =` and change it:

```python
import matplotlib.pyplot as plt
import numpy as np
from scipy.spatial.transform import Rotation

# from pvlib.tracking import calc_surface_orientation
from pvlib.tools import acosd, cosd, sind

def _unit_normal(axis_azimuth, axis_tilt, theta):
    theta = np.asarray(theta)

    cA, sA = cosd(-axis_azimuth), sind(-axis_azimuth)
    cT, sT = cosd(-axis_tilt), sind(-axis_tilt)

    cTh = cosd(theta)
    sTh = sind(theta)

    x = sA * sT * cTh + cA * sTh
    y = sA * sTh - cA * sT * cTh
    z = cT * cTh

    result = np.column_stack((x, y, z))

    return result

def calc_surface_orientation(tracker_theta, axis_tilt=0, axis_azimuth=0):
    with np.errstate(invalid="ignore", divide="ignore"):
        surface_tilt = acosd(cosd(tracker_theta) * cosd(axis_tilt))

    # unit normal to rotated tracker surface
    unit_normal = _unit_normal(axis_azimuth, axis_tilt, tracker_theta)

    # deviate from [1] to allow for negative tilt.
    # project unit_normal to x-y plane to calculate azimuth
    surface_azimuth = np.degrees(np.arctan2(unit_normal[:, 0], unit_normal[:, 1]))
    # constrain angles to [0, 360)
    surface_azimuth = np.mod(surface_azimuth, 360.0)

    surface_azimuth = np.where(surface_tilt == 0.0, axis_azimuth, surface_azimuth)

    out = {
        "surface_tilt": surface_tilt,
        "surface_azimuth": surface_azimuth,
    }
    if hasattr(tracker_theta, "index"):
        out = pd.DataFrame(out)
    return out

plt.interactive(True)

# Create rectangle vertices (10 in x, 20 in y)
x = np.array([-5, 5, 5, -5, -5])  # Close the loop by repeating first point
y = np.array([-10, -10, 10, 10, -10])
z = np.zeros_like(x)

# Stack coordinates
points = np.vstack([x, y, z])

theta = 0
axis_azimuth = 180
axis_tilt = 0
surface_orientation = calc_surface_orientation(
    theta, axis_tilt=axis_tilt, axis_azimuth=axis_azimuth
)
surface_azimuth = surface_orientation["surface_azimuth"][0]
surface_tilt = surface_orientation["surface_tilt"]
# Apply rotations
# surface_azimuth = 90
# surface_tilt = 30
rot = Rotation.from_euler(
    "ZX", [-surface_azimuth, -surface_tilt], degrees=True
)

points_rot1 = rot.apply(points.T).T

# Extract rotated coordinates
x_rot, y_rot, z_rot = points_rot1

# Create the figure and 3D axis
fig = plt.figure(figsize=(10, 8))
ax = fig.add_subplot(111, projection="3d")

# Plot wireframe (connect the vertices in order)
ax.plot(x_rot, y_rot, z_rot, "b-", linewidth=2, label="Rectangle wireframe")

# The rectangle is defined by 4 vertices, we've already closed the loop
# Additional lines for the rectangle structure
for i in range(4):
    ax.plot(
        [x_rot[i], x_rot[(i + 1) % 4]],
        [y_rot[i], y_rot[(i + 1) % 4]],
        [z_rot[i], z_rot[(i + 1) % 4]],
        "b-",
        linewidth=2,
    )

# Set labels
ax.set_xlabel("> to East")
ax.set_ylabel("< to North")

# Set equal aspect ratio
ax.set_box_aspect([1, 1, 1])

# Position camera to look towards Y+ direction (North)
# Matplotlib convention differs from pvlib's
ax.view_init(elev=30, azim=-90)

# Set limits to show the rectangle properly
max_range = 20

ax.set_xlim(-max_range / 2 - 5, max_range / 2 + 5)
ax.set_ylim(-max_range / 2 - 5, max_range / 2 + 5)
ax.set_zlim(-max_range / 2 - 5, max_range / 2 + 5)

ax.set_title(
    f"Rectangle 10x20\nRotated: {surface_azimuth=:.0f}°, {surface_tilt=:.0f}°\n"
    "Camera facing North"
)

plt.show(block=True)
```

</p>
</details> 

Disclaimer: I don't think anybody should ever take this approach for drawing 3D PV scenes, but I can't guarantee nobody will. Note the code above is only valid for `axis_tilt=0`. I haven't had the energy to fix it for non-zero values.

This example is just for the purpose of illustrating that picking this value may introduce unwanted discontinuities into the dataflow.