pvlib/pvlib-python #1980: Allow arbitrary IAM function in pvlib.iam.marion_diffuse, and possibly improve horizon integral computation.

markcampanelli, 2024-02-24:
**Is your feature request related to a problem? Please describe.**
I am working on an application using pvlib.iam.marion_diffuse, and I would like to inject my own IAM function as a parameter to this pvlib.iam.marion_diffuse.

**Describe the solution you'd like**
I would like to define the common interface for the existing IAM function choices fo pvlib.iam.marion_diffuse, then allow any function meeting this interface (or Protocol, in Python speak) to be passable to pvlib.iam.marion_diffuse.

**Describe alternatives you've considered**
Do this on my own and not try to port it back to pvlib-python.

**Additional context**
If my understanding is correct, there is also a potential simplification to the Marion integral for the horizon component, which for models like Perez is considered to be concentrated in the 2D horizon ring.

kandersolar, 2024-02-24:
Note that `marion_diffuse` is a simple wrapper around [marion_integrate](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.iam.marion_integrate.html), which takes an arbitrary function `f(aoi)` and integrates it across a region.  So it is already possible to use custom IAM functions, although it's somewhat cumbersome. Some brief discussion of this idea is in the original PR: https://github.com/pvlib/pvlib-python/pull/984#discussion_r440505518

I think it would be a straightforward modification to allow `marion_diffuse` to take functions as well as the current model names.

markcampanelli, 2024-02-25:
@kandersolar Thanks for the background. I see in the history that your first inclination was apparently to make `marion_diffuse` simply take any callable IAM function :). Given the way `pvlib-python` tends to prefer ease-of-use over extensibility, I think relying on the lower level `marion_integrate` is probably sufficient for this repo. I'm going to close this.

closed 2024-02-25 by markcampanelli

markcampanelli, 2024-04-15:
For posterity (using `pvlib` 0.10.4), here is a computational comparison of the three correction factors from the Marion model using the `physical` setting in `pvlib.iam.marion_diffuse` vs. using `pvlib.iam.marion_integrate` with the "custom" IAM model derived from a PCHIP interpolation to IEC 61853-2 data (see graph).

`pvlib.iam.marion_diffuse` using `physical` model
{'ground': 0.7998463278034974,
 'horizon': 0.8975832800966072,
 'sky': 0.9587871231479588}

`pvlib.iam.marion_integrate` using interpolated data IEC 61853-2:
{'ground': 0.8312793473238048,
 'horizon': 0.9199299978256542,
 'sky': 0.9691309027924638}

Thanks again @kandersolar for suggesting the workaround :).

[image]