pvlib/pvlib-python #1984: First Solar CEC modules are inferred to be CIGS instead of CdTe

kandersolar, 2024-03-06:
**Describe the bug**
Some First Solar modules are incorrectly inferred to be CIGS instead of CdTe.  This is because the CEC database we currently include in pvlib lists these modules as "Thin Film", which `pvlib.pvsystem.Array` maps to CIGS:

https://github.com/pvlib/pvlib-python/blob/923e025dd32a05ad7c9beaa442d2b83a7be53731/pvlib/pvsystem.py#L1024-L1031

**To Reproduce**
```python
import pvlib

cec_modules = pvlib.pvsystem.retrieve_sam('cecmod')
fslr_series_4 = cec_modules['First_Solar__Inc__FS_4117_3']

array = pvlib.pvsystem.Array(pvlib.pvsystem.FixedMount(), module_parameters=fslr_series_4)
print(array._infer_cell_type())  # shows cigs instead of cdte
```

That means that ModelChains using the first solar spectral correction model will incorrectly use CIGS parameters instead of CdTe parameters for these modules.

**Expected behavior**
First Solar modules should be inferred as CdTe, not CIGS.

**Versions:**
 - ``pvlib.__version__``: 0.10.3

**Additional context**
These modules are still listed as `Thin Film` in the latest version of the CEC table from SAM: https://github.com/NREL/SAM/blob/patch/deploy/libraries/CEC%20Modules.csv, so updating our table won't fix this problem.

cwhanse, 2024-03-13:
Looks like a long-standing issue in the CEC data, where "Technology" has been "Thin film" for First Solar products from Series 4 and earlier since 2018 (archive [here](https://github.com/NREL/SAM/tree/patch/samples/CEC%20Module%20and%20Inverter%20Libraries/CEC%20Modules)). 

Maybe this could be better addressed in the processing of CEC data to the SAM file? @janinefreeman

janinefreeman, 2024-03-13:
@cpaulgilman FYI. I just took a look at this in SAM, and it appears that we may be having a similar problem- module technologies listed as "thin film" in the database are getting assigned to a default type for `6par_celltech`. At a cursory glance, that input seems to only inform the initial guesses for some of the parameter solution functions in `6par_solve.h`, so I'm not sure it's affecting SAM outputs per se. Would it be possible to map FirstSolar modules to cdte in our processing of the CEC data? I know we've had problems trying to correct things based on names before...

cpaulgilman, 2024-03-13:
> Would it be possible to map FirstSolar modules to cdte in our processing of the CEC data? I know we've had problems trying to correct things based on names before...

@janinfreeman In SSC, `6par_celltech` is an input to the 6parsolve parameter solver module, not to the module model itself, so I agree that does not affect the module model outputs. 

In SAM, the module model does not use the "Technology" value shown in the module library for the "CEC Performance Model with Module Database" option on the Module page. The cell type (`6par_celltech`) is a user-provided input for the "CEC Peformance Model with User Entered Specifications" option. When users use the **Copy specifications...** button for the User Entered Specifications option to copy data from the Module Database option's library for a module with Technology = Thin Film, the SAM UI sets `6parcelltech = 2` (CdTe). So, for that specific workflow, the cell type for all First Solar modules does get set CdTe.

Here's a screenshot of the LK script from the SAM UI callback for the CEC Performance Model with User Entered Specifications input form:

[image]

Finally, in the module library for SAM 2023.12.17, First Solar modules with model numbers between FS-267 and FS-4122 are listed as "Thin Film." Model numbers FS-6385 to FS 7640 are listed as "CdTe," reflecting the way they are identified in the CEC Excel workbook.  I'm hesitant to modify our library scripts to treat a particular manufacturer, or range of model numbers within a manufacturer, differently than others. 

The scripts and files we use to create the SAM module libraries are here: https://github.com/NREL/SAM/tree/patch/samples/CEC%20Module%20and%20Inverter%20Libraries/CEC%20Modules

cwhanse, 2026-11-03:
Since this attracted an AI PR, I propose that the way forward to close the issue is to NOT infer coefficients for the First Solar spectral model when the CEC Technology is "Thin Film". A user can instead 1) overwrite the CEC Technology to specify "CdTe", or 2) add the First Solar spectral model parameters to module_parameters.