pvlib/pvlib-python #1986: Whether to consider supporting pv power generation in mountainous scenarios? The mountainous terrain can be an irregular triangular network.

xiuweihan, 2024-03-08:
Whether to consider supporting pv power generation in mountainous scenarios? 

The mountainous terrain can be an irregular triangular network(TIN) or elevation point data.

kandersolar, 2024-03-08:
Here is an example of modeling the effect of far-away mountains blocking the sun as it approaches the horizon: https://pvlib-python.readthedocs.io/en/stable/gallery/shading/plot_simple_irradiance_adjustment_for_horizon_shading.html

The effect of different module orientations due to terrain variation within the array could be modeled with pvlib, but the effect of row-to-row shading within the array cannot. 

Supporting terrain meshes or DEMs might be outside the scope of pvlib, but it could be a separate project that interoperates with pvlib.  But it's not a small undertaking, and someone would have to have the time and motivation to do it.

kurt-rhee, 2024-11-04:
@kandersolar do you think that this would warrant a function in pvlib?

cwhanse, 2024-11-04:
> this would warrant a function in pvlib?

What would the function do?