pvlib/pvlib-python #1992: v0.10.4 release planning

kandersolar, 2024-03-12:
Release v0.10.4 is scheduled for this Friday.  Here is a proposed release plan for discussion:

Include in v0.10.4:
- #1689, #1690, #1962
- #1944 
- #1952, #1985
- #1979, pending agreement that the method is sufficiently well-defined in the reference

Defer to the future:
- #1708
- #1763 
- #1781 
- #1893

adriesse, 2024-03-13:
It looks like I haven't made the contributor list yet.  My gut feeling is I should be on it based on a variety of input.

kandersolar, 2024-03-13:
Yep, we are definitely sloppy about making sure each PR records all of its contributors.  Instead, we have a step in the [release procedure](https://github.com/pvlib/pvlib-python/wiki/Release-procedures#final-release) to add the names of any missed reviewers and such, so the "finalize release" PR will get you on there :)

closed 2024-03-19 by kandersolar

adriesse, 2024-03-20:
Thanks for keeping things moving, @kandersolar!

PR #1994 (2024-03-19): Finalize 0.10.4
 - [x] Closes #1992
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` o [... cut]