pvlib/pvlib-python #1993: Single Diode Model error when low effective irradiance (pvlib==0.10.2)

OnairSandra, 2024-03-13:
pvlib usage questions can be asked on [Stack Overflow](http://stackoverflow.com) and tagged with the [pvlib](http://stackoverflow.com/questions/tagged/pvlib) tag.

The [pvlib-python google group](https://groups.google.com/forum/#!forum/pvlib-python) is used for discussing various topics of interest to the pvlib-python community. We also make new version announcements on the google group.

I get a ValueError when using single diode model with lambertw method.
                p_mp, v_mp = _golden_sect_DataFrame(params, 0., v_oc * 1.14, _pwr_optfcn)
                raise ValueError('upper >= lower is required')

I've seen some issues that may be related, because newton method doesn't raise any error, and I've some very low irradiance values:
  https://github.com/pvlib/pvlib-python/issues/1673
  https://github.com/pvlib/pvlib-python/issues/1843

Debugging lambertw method, I found the following:
#Set small elements <0 in v_oc to 0
    if isinstance(v_oc, np.ndarray):
        v_oc[(v_oc < 0) & (v_oc > -1e-12)] = 0.

In my case, there are some v_oc values under 0, but no so close to zero as -1e-12
v_oc[(v_oc < 0)]  =  array([-0.00024414, -0.00097656, -0.00195312, -0.00390625, -0.00390625])

#Questions:
Is there any reason to not filter every negative  value in v_oc?
May I use newton method to avoid this error?

#Environment:
Python version: 3.10
pvlib version: 0.10.2

cwhanse, 2024-03-13:
@OnairSandra when we've looked at similar reports, the root cause has been very low (e.g. 1e-9) values for effective irradiance. The solution has been to filter the effective irradiance before calling the `calc_params` and `singlediode` functions.

To know for certain you would need to provide code and some data.

Or you can try `singlediode(...., method='newton'...)` which finds the maximum power point with a different method.

cwhanse, 2024-03-13:
If we have a reason to not filter, it's to avoid covering up numerical bugs by replacing bad values with 0.