pvlib/pvlib-python #2003: Snow albedo change model

NormanJost, 2024-04-03:
Hi everyone,

Dan Riley and Chris Pike developed this model to calculate albedo change from snow depths and temperatures.
Here is the reference to their conference proceeding with a full paper currently in submission.
[10.1109/PVSC48317.2022.9938897](https://doi.org/10.1109/PVSC48317.2022.9938897
)

The model is already making it's way into pvlib-matlab. I'm currently working on a python version to be included soon.

AdamRJensen, 2024-04-07:
I can't access the paper, could you type up the equation and post it here?

adriesse, 2024-04-12:
> I can't access the paper, could you type up the equation and post it here?

Or post a preprint somewhere?

DanRiley, 2024-04-16:
I'll see what I can do about getting a preprint available.

DanRiley, 2024-04-16:
> > I can't access the paper, could you type up the equation and post it here?
> 
> Or post a preprint somewhere?

I'll see what I can do about getting a preprint available.

DanRiley, 2024-10-22:
Our paper was published in MDPI _Solar_. Here's the citation.

Pike_ C, Riley D, Toal H, Burnham L. Presenting a Model to Predict Changing Snow Albedo for Improving Photovoltaic Performance Simulation. Solar. 2024; 4(3):422-439. https://doi.org/10.3390/solar4030019