pvlib/pvlib-python #2008: Codecov CI error uploading report

echedey-ls, 2024-04-12:
**Describe the bug**
Originally from [#2006 (comment)](https://github.com/pvlib/pvlib-python/pull/2006#issuecomment-2051394258), but also seen at #2007.
The reported error is related to upload.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any PR updated on the last days and see the failing workflow.
    - For example: https://github.com/pvlib/pvlib-python/actions/runs/8659882128/job/23746657398?pr=2006#step:10:102

**Expected behavior**
To be uploaded correctly.

**Screenshots**
[image]

**Versions:**
N/A

**Additional context**
These links may be useful:
* Long discussion at https://github.com/codecov/codecov-action/issues/598
* What seems to be the problem and the fix, instructions, https://community.codecov.com/t/upload-issues-unable-to-locate-build-via-github-actions-api/3954

Pinging @pvlib/maintainers, I can't be of help with this issue.

kandersolar, 2024-04-29:
Thanks @echedey-ls.  Looks like codecov very recently made it easier for us to solve this problem: PRs from forks no longer require a codecov token to upload coverage data.  It seems to be working, at least in #2022.

However, coverage also gets uploaded when new commits land on `main`.  From the codecov docs, I think these uploads will not be able to use the "tokenless" route available for PRs.  For that, I have added a github secret with a codecov upload token.  

With those two changes, I think this problem will be solved.  I will proceed with merging #2022 and see if things work as expected on `main`.

closed 2024-04-29 by kandersolar

kandersolar, 2024-04-29:
Looks like everything is working after the second PR (#2024): https://app.codecov.io/gh/pvlib/pvlib-python/commit/25ec2968d97af1a0ad9814dc559b7ebaaaa619e2

PR uploads may still randomly fail; I don't think there's much we can do about that, at least for now.  Re-triggering failed jobs continues to be the workaround.

Thanks @echedey-ls for making an issue :)

PR #2022 (2024-04-29): Update codecov CI configuration to fix PR coverage uploads

 - [x] Closes #2008
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to th [... cut]