pvlib/pvlib-python #2010: Update Comparison with PVLIB MATLAB

AdamRJensen, 2024-04-12:
The description contained in the [Comparison with MATALB](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/user_guide/comparison_pvlib_matlab.rst) section is very outdated.

For example, it is stated that:
> We'd all like to keep the core functionality of the Python and MATLAB projects synchronized,

This makes it sound like the MATLAB and python version are at somewhat similar stages, whereas my impression is that the MATLAB version receives very little maintenance and development and is thus far behind pvlib-python.

@cwhanse are you up for updating this description?

Alternatively, do we even want an entire page dedicated to pvlib MATLAB? Perhaps this section could be replaced by an additional sentence or two in the [History and acknowledgement](https://pvlib-python.readthedocs.io/en/stable/user_guide/package_overview.html) section?

cwhanse, 2024-04-12:
We should update that. I'll draft a PR. I may end up removing that page and adding a sentence as you suggest.

AdamRJensen, 2024-04-16:
It'd also be nice to remove the MATLAB comment in the [contributing section](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/contributing.rst):
> If you have MATLAB experience, you can help us keep pvlib-python up to date with PVLIB_MATLAB or help us develop common unit tests. For more, see https://github.com/pvlib/pvlib-python/issues/2 and https://github.com/pvlib/pvlib-python/issues/3.

PR #2012 (2024-04-16): move Matlab references to history
 - [X] Closes #2010
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~~[ ] Tests added~~
 - ~~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue wit [... cut]
closed 2024-04-16 by kandersolar