pvlib/pvlib-python #2017: Inferring spectral model returns 'first_solar' when not expected

cwhanse, 2024-04-25:
A user reports running a model chain with no spectral model specified and with CEC module parameters. The logic for `ModelChain.infer_spectral_model` returns `first_solar_spectral_loss`, which expects `'precipitable_water'` to be in the weather input, at which point the user's simulation fails.

**To Reproduce**
[Code](https://groups.google.com/g/pvlib-python/c/3si6K8EVgN4/m/roKyvNwXBQAJ?utm_medium=email&utm_source=footer) posted by user to reproduce the error.
```
import pandas as pd
import pvlib

from pvlib import clearsky, atmosphere, solarposition
from pvlib.modelchain import ModelChain
from pvlib.location import Location
from pvlib.pvsystem import PVSystem, Array, SingleAxisTrackerMount, AbstractMount, FixedMount
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS

location = Location(latitude = 40.30117520975112, longitude = -3.6973221576701967, tz = "Europe/Madrid", altitude = 660, name = 'Ormazabal Getafe')
modules = pvlib.pvsystem.retrieve_sam('CECMod')
inverters = pvlib.pvsystem.retrieve_sam('CECInverter')
module = modules['JA_Solar_JAP72S01_330_SC']
inverter = inverters['SMA_America__STP_33_US_41__480V_']
temp_par = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']
system = PVSystem(surface_tilt = 19, surface_azimuth = 170, module_parameters = module, inverter_parameters = inverter,
                  temperature_model_parameters = temp_par, modules_per_string = 17, strings_per_inverter = 6)
modelchain = ModelChain(system, location, aoi_model = 'physical')

times = pd.date_range(start = '2023-06-01', end = '2023-12-15', freq = 'h', tz = location.tz)
solar_position = location.get_solarposition(times)
clearsky = location.get_clearsky(times)

modelchain.run_model(clearsky)

```
**Expected behavior**
While the CEC database has enough information (cell technology) to select parameters for the First Solar spectral model, doing so creates an expectation that the user has the inputs for that model. We can't check weather data at the time that `ModelChain.spectral_model` is set. I am inclined to think that the `elif` [here](https://github.com/pvlib/pvlib-python/blob/1d854e5a83543de34337a63356efb9c8dd3d859a/pvlib/modelchain.py#L1043) should go away.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions:**
 - ``pvlib.__version__``:  0.10.5

mikofski, 2024-04-27:
Also reported in stack overflow: https://stackoverflow.com/questions/78382877/keyerror-while-plotting-with-pvlib

echedey-ls, 2024-10-02:
Is anybody working on this? I may be able to fix it next week.

cwhanse, 2024-10-02:
I am not working on it - you are welcome to fix it.

PR #2253 (2025-03-14): [BUG]: don't infer spectral model in ModelChain
 - [x] Closes #2017
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub I [... cut]
commit b4916e11 (2025-03-14): [BUG]: don't infer spectral model in ModelChain (#2253)

* Fix #2017

* rm TODOs

* cwhanse review

* typo

* pep8 compliance

* ups

* Update modelchain.py

* Infer a string, key for setter

* pep8 linting

* Remove delayed-inference code - codecov warning

* default to no_loss

* infer_spectral_model tweak -> return list if possible

* remove infer_spectral_model

* lint

* review

* remove reference to infer_spectral_model method

---------
closed 2025-03-14 by kandersolar