pvlib/pvlib-python #2018: Using updated Sam datase

apct69, 2024-04-26:
pvlib usage questions can be asked on [Stack Overflow](http://stackoverflow.com) and tagged with the [pvlib](http://stackoverflow.com/questions/tagged/pvlib) tag.

The [pvlib-python google group](https://groups.google.com/forum/#!forum/pvlib-python) is used for discussing various topics of interest to the pvlib-python community. We also make new version announcements on the google group.

echedey-ls, 2024-04-26:
Hmm, from looking at the code, when `name=` is given, it ignores `path` silently. Just drop the first argument.

That code line should be `retrieve_sam(path="https://raw.githubusercontent.com/NREL/SAM/develop/deploy/libraries/CEC%20Modules.csv")`

@pvlib/maintainers, I think both the documentation and code need a bit of rework to avoid confusion. In fact, I think this should be a keyword-only function.

echedey-ls, 2024-04-26:
Btw @apct69 , you were looking at v0.4.3 documentation (from 2016). We are now at v0.10.4 - there are a few things that have changed since then.

Please follow this link: https://pvlib-python.readthedocs.io/en/stable/

apct69, 2024-04-26:
Thanks for the fast response, i will be able to choose my favorite panels then :)

I will be looking to the new version of the documentation to learn the use of PVLIB!

PR #2020 (2024-05-01): Fix silently ignoring file path in `pvsystem.retrieve_sam` when `name` is provided
 - [x] Closes #2018 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all chan [... cut]
closed 2024-04-30 by apct69