pvlib/pvlib-python #2042: Unexpected behavior of get_total_irradiance function in pvlib

amaccarini, 2024-05-07:
I am developing a simple code to calculate the solar radiation on a tilted surface using the function `get_total_irradiance`. Here the code:

      from pvlib import location
      from pvlib import irradiance
      import pandas as pd
      from matplotlib import pyplot as plt
      import pvlib

      epw_file = "/USA_IL_Chicago-OHare.Intl.AP.725300_TMY3.epw"  
      epw_data = pvlib.iotools.read_epw(epw_file)

      lat=epw_data[1]['latitude']
      lon=epw_data[1]['longitude']
      ghi=epw_data[0]['ghi'].tz_localize(None).reset_index(drop=True)
      dni=epw_data[0]['dni'].tz_localize(None).reset_index(drop=True)
      dhi=epw_data[0]['dhi'].tz_localize(None).reset_index(drop=True)

      loc = location.Location(lat,lon)
      times = pd.date_range(start='01-01-2001', end='12-31-2001 23:00', freq='1H')
      sp = loc.get_solarposition(times)

     solar_zenith=sp['apparent_zenith'].reset_index(drop=True)
     solar_azimuth=sp['azimuth'].reset_index(drop=True)

     POA_irradianceN = irradiance.get_total_irradiance(surface_tilt=90, surface_azimuth=0, 
     solar_zenith=solar_zenith, solar_azimuth=solar_azimuth, dni=dni, ghi=ghi, dhi=dhi)

     POA_irradianceE = irradiance.get_total_irradiance(surface_tilt=90, surface_azimuth=90, 
     solar_zenith=solar_zenith, solar_azimuth=solar_azimuth, dni=dni, ghi=ghi, dhi=dhi)

     POA_irradianceS = irradiance.get_total_irradiance(surface_tilt=90, surface_azimuth=180, 
     solar_zenith=solar_zenith, solar_azimuth=solar_azimuth, dni=dni, ghi=ghi, dhi=dhi)

     POA_irradianceW = irradiance.get_total_irradiance(surface_tilt=90, surface_azimuth=270, 
     solar_zenith=solar_zenith, solar_azimuth=solar_azimuth, dni=dni, ghi=ghi, dhi=dhi)`

According to the documentation, the surface_azimuth of a surface oriented to South should be 180°. However, it looks like the South corresponds to 90°. I tried with a different climate file from the US (USA_CA_San.Francisco.Intl.AP.724940_TMY3) and I get the same issue. However, when I try with climate files from Europe (for example ITA_Venezia-Tessera.161050_IGDG), everything works smoothly. Could it be related to the fact that the US files are based on TMY3? Are you familiar with this behavior? Any insights or thoughts on what might be causing it?

kandersolar, 2024-05-07:
These two lines are a problem:
```python
  times = pd.date_range(start='01-01-2001', end='12-31-2001 23:00', freq='1H')
  sp = loc.get_solarposition(times)
```

The first line creates a timezone-"naive" index (timestamps not associated with a time zone).  In that case, pvlib calculates solar positions assuming they are UTC.  Since the irradiance values returned by read_epw are timestamped to local time, there a timing mismatch of several hours between the solar positions and the irradiance values.

To prevent that issue, calculate the solar positions using a timezone-aware index such as the one that is returned by read_epw.  In fact it is better to use a single timezone-aware index throughout your code, both to prevent issues such as this one and to avoid any need for the `drop_index()` and `tz_localize(None)` in your current code.

amaccarini, 2024-05-07:
Thanks for the help @kandersolar! I edited my code as:

```python
times = pd.date_range(start='01-01-2001', end='12-31-2001 23:00', freq='1H', tz='America/Chicago')
sp = loc.get_solarposition(times)
```

It works well now. I actually tried to remove `drop_index()` and `tz_localize(None)` but that created some problems. So I will keep them in the code. Now I just have to figure out a way to convert the `tz` read by the epw (which is a float) into a string to be read by the `pd.date_range`.

kandersolar, 2024-05-08:
The DataFrame returned by `read_epw` should have a suitable timezone already within its datetime index.  So you should be able to do something like this:

```python
df, metadata = read_epw(...)
sp = loc.get_solarposition(df.index)  # use df.index instead of manually creating a "times" index
solar_zenith = sp['apparent_zenith']
solar_azimuth = sp['azimuth']
POA_irradianceN = irradiance.get_total_irradiance(surface_tilt=90, surface_azimuth=0, 
  solar_zenith=solar_zenith, solar_azimuth=solar_azimuth, dni=dni, ghi=ghi, dhi=dhi)
```

Anyway, since there is no problem with pvlib, I'm going mark this as "closed".

closed 2024-05-08 by kandersolar