pvlib/pvlib-python #2043: Discontinuous Diffuse Sky Irradiance from Perez Model

jmobrecht, 2024-05-09:
**Describe the bug**
Using PVLib functions, I calculate a clear sky irradiance and find small ~1% discontinuities originating from the Perez model.  I have isolated the issue in the source code and can correct the issue with only a few additional lines and without any cost in computational time.

**To Reproduce**
Steps to reproduce the behavior:
1. Calculate solar position using pvlib.solar_position.get_solarposition
2. Calculate air pressure pvlib.atmosphere.alt2pres
3. Calculate airmass pvlib.atmosphere.get_relative_airmass
4. Calculate Linke Turbidity pvlib.clearsky.lookup_linke_turbidity
5. Calculate DNI_extra pvlib.irradiance.get_extra_radiation
6. Calculate GHI, DNI, DHI from pvlib.clearsky.ineichen

From the above inputs, I calculate the diffuse sky irradiance using the pvlib.irradiance.perez model.  The output results show a sharp discontinuity, which arises from calculation of the F1 & F2 coefficients in the 'perez' function.  When an interpolation is used, the discontinuity is eliminated.

**Expected behavior**
Attached is an image showing the discontinuity of the diffuse sky irradiance from PVLib (blue) and the irradiance from my solution (JMO - red).
[image]

**Solution**

**Comparison**
Code runs just as fast as original --- no computational cost to solution.

**Final Note**
Just spoke with PVLib maintainer Adam who told me that this solution is already solved (https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.perez_driesse.html)... regardless I'm posting my solution for anyone who wants to use it.

AdamRJensen, 2024-05-13:
Thanks for bringing this up @jmobrecht.

Would it be an idea to add a warning admonition on the Perez documentation page, stating that there are small discontinuities and point to the Perez-Driesse model?

kandersolar, 2024-06-14:
These minor discontinuities in the output of the Perez model are a frequent source of surprise.  I think a brief explanatory note in the docstring is a good idea.

PR #2098 (2024-06-21): Add hint about Perez discontinuities

 - [x] Closes #2043
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Pull request is nearly complete and ready for detailed review.
 - [x] Maintainer: Appropriate GitHub Labels (including `remote-data`) and Milestone are assigned to the Pull Request and linked Issue.

This PR adds a Hint section to the ``irradiance.get_sky_diffuse`` function stating that the Perez model output is discontinuous and suggests using the Perez-Driesse model for applications where this may be an issue [... cut]
closed 2024-06-21 by kandersolar