pvlib/pvlib-python #2044: Dynamic Snow Loss Model

kylefmacdonald, 2024-05-10:
**Is your feature request related to a problem? Please describe.**
Snow loss modelling is difficult and currently the only model available on pvlib at an hourly timestep is the Marion model: 

https://pvlib-python.readthedocs.io/en/v0.10.4/reference/generated/pvlib.snow.coverage_nrel.html

However this model does not account for the movement of a tracker and the snow pile that may accumulate.

**Describe the solution you'd like**
In 2018, Defne Gun, Mike Anderson, Greg Kimball, and Ben Bourne developed a dynamic snow loss model:

https://gregorykimball.me/wp-content/uploads/2018/07/mewcpec1079_0614162840_snow_2018.pdf

A few years ago I translated the attached pdf to python. My solution would be to contribute this code to pvlib so the industry may model snow losses accounted for the dynamic nature of Horizontal Single Axis Tracker (HSAT) systems. I wrote this attached code 2 years ago when I was relatively new to python. I would appreciate any feedback and advice you have however in the meantime I will also review my code as I am sure it can be improved. After an initial look I will start by refactoring to functional programming (I can also refactor to OOP if that is what you would prefer). In addition I will also update the code to follow the pvlib guidelines.

The output of the attached code is the adjusted POA irradiance due to snow soiling, however I am open to suggestions on the output and how the losses should be applied. I have also created a script that combines the PVSIM + Ryberg model however that can be a future issue.

[pvsim.zip](https://github.com/pvlib/pvlib-python/files/15279792/pvsim.zip)

**Describe alternatives you've considered**
No other dynamic model currently exists in pvlib, I am simply using a model that has already been developed and released publicly.

Thank you in advance for the guidance and help on this, looking forward to contributing!

echedey-ls, 2024-05-10:
Now that's what I call well commented code! The functional programming is perfectly fine. OOP can be tackled down in another PR.

Just a heads up, that's a pretty big model, so be patient to get it all working and reviewed. In any case, that's gonna be a nice contribution!

Recommended readings and tips:
- https://pvlib-python.readthedocs.io/en/stable/contributing.html
- rST cheatsheet, to redact documentation https://bashtage.github.io/sphinx-material/rst-cheatsheet/rst-cheatsheet.html
- Develop in the lowest python version supported (now it is 3.8), so things don't fail unexpectedly
- Install the package in editable mode at least with the `test` optional packages in a virtual environment
  ```bash
  python -m venv venv
  venv/Scripts/activate or venv/bin/activate
  python -m install -e .[test]
  ```

kylefmacdonald, 2024-05-10:
Hey Luis,

Thank you for the tips! Once I functionalize the code and implement your tips should I reply to the issue with the new attached code?

Thanks,

Kyle

  *   https://pvlib-python.readthedocs.io/en/stable/contributing.html

kylefmacdonald, 2024-05-10:
Hey Luis,

Can you please take this issue down for now. I need to get permission from my work before I proceed. Once I get permission I will make a new issue and proceed.

Thank you,

Kyle

  *   https://pvlib-python.readthedocs.io/en/stable/contributing.html

echedey-ls, 2024-05-10:
Feel free to do what you need, although I'd leave this open at least as a possible addition!

In any case, if you go further with the code, opening a pull request is possibly a better fit!

(Btw, I'm just a contributor, I wouldn't be able to close your issue)

kylefmacdonald, 2024-05-10:
Alright sounds good I’ll keep it open until I get confirmation and we can go from there. If I get the green light I’ll create a pull request. What directory would be best?

echedey-ls, 2024-05-11:
It would be too boring if I told you everything that must be done. It's a requirement to figure out some of the changes ;). What's more, isn't open source just another form of human creativity? I'd say it's the finest one.

Seriously, lurk over the repo, it will also help you be confident with it.

kylefmacdonald, 2024-05-13:
Hey Luis,

I would like to take this issue down for now. Could you please forward this request to a maintainer so this issue can be deleted. Once I get permission, I will create a pull request as you suggested. However, until then I would like to delete this issue.

Thank you,

Kyle

cwhanse, 2024-05-14:
Kyle,

I'm confused - why would you like the issue deleted? Normally, we leave issues open until corresponding pull requests are merged. Or the originator (you) should be able to close the issue, if you want, but I don't see the need. Adding that snow loss model improvement seems worthwhile to me.

Cliff

kylefmacdonald, 2024-05-14:
Hey Chris,

Thank you for following up. I agree, I do think it would be a notable project. However I am waiting for approval from my company to see if I can work on this outside of work before I proceed which is why I wanted to delete the issue. I misunderstood what creating an issue meant. However I will leave it up while I wait for an answer from my company.

Thank you,

Kyle

cwhanse, 2024-05-14:
I understand. Just to be clear, opening the issue doesn't obligate you to provide code.