pvlib/pvlib-python #2045: NREL mismatch losses model due to non-uniform irradiance of cells

echedey-ls, 2024-05-10:
**Is your feature request related to a problem? Please describe.**
Add a the model described [here](https://research-hub.nrel.gov/en/publications/bifacial-pv-system-mismatch-loss-estimation-and-parameterization-). Eq. 7 is the relevant one.

**Describe the solution you'd like**
A function with it's input "relative mean absolute error" in `pvlib.pvsystem` (other losses models live here). Documentation/example would add the extra mathematical layers since they can vary. In the upcoming PR I will show cases of this.

**Describe alternatives you've considered**
Including the RMAD calculation in `pvlib.tools`, but I think it doesn't make much sense to add such a mathematical function. In the example it can be supplied.

**Additional context**
I'm adding an example case with data from an HSAT mock-up my thesis tutors made.

kandersolar, 2024-05-13:
See also #1541 (should we close this one as a duplicate?).  The cited reference there (Prog in PV) is probably better than the EU PVSEC proceedings.

echedey-ls, 2024-05-13:
Definitely. Thanks for the reference @kandersolar !

closed 2024-05-13 by echedey-ls