pvlib/pvlib-python #2047: Photosynthetically active radiation decomposition model

echedey-ls, 2024-05-12:
Following my GSoC proposal, there are two models I plan to contribute regarding Photosynthetically Active Radiation, PAR. This is of great importance in the design of AgriVoltaic systems, since PAR is one of the metrics used for the simulation of crops productivity.

I pretend this is the first one due to simplicity. The model takes an input of the global irradiance diffuse fraction and the solar elevation, and outputs the fraction of diffuse PAR. See Eq. (1) in [2]. It is a derivation of equations (9) & (10) in [1]. This is known as the Spitters relationship in [2].

- [1] C. J. T. Spitters, H. A. J. M. Toussaint, and J. Goudriaan,
       'Separating the diffuse and direct component of global radiation and its
       implications for modeling canopy photosynthesis Part I. Components of
       incoming radiation', Agricultural and Forest Meteorology, vol. 38,
       no. 1, pp. 217-229, Oct. 1986, [:doi:`10.1016/0168-1923(86)90060-2`.](https://doi.org/10.1016/0168-1923(86)90060-2)
- [2] S. Ma Lu et al., 'Photosynthetically active radiation decomposition
       models for agrivoltaic systems applications', Solar Energy, vol. 244,
       pp. 536-549, Sep. 2022, [:doi:`10.1016/j.solener.2022.05.046`.](https://doi.org/10.1016/j.solener.2022.05.046)

**Describe the solution you'd like**
A function in a new module, presumably `pvlib.par`, that has the function `spitters_relationship` with the described inputs and output.

**Describe alternatives you've considered**
Other names for the module `pvlib.agrivoltaic[s]` or trying to fit it into current structure until new things can be included and then arranged in the future. Take into account moving things requires deprecations.
Other names for the function, may be `par_diffuse_fraction_from_global` or so. However, I think an author-named model is better in this case. There may be more models about that, like the modified by Gu et al., Eq. (3) in [2].

As always, I'm open to other alternatives.

**Additional context**
I will open a PR in short as an initial proposal.

Ask, discuss and suggest freely.

kandersolar, 2024-05-13:
I am a complete non-expert in agriPV, but my initial reaction is that PAR is too narrow a modeling scope to warrant its own `pvlib.par` module.  I could potentially be convinced that a broader `pvlib.agrivoltaics` module is a good idea, but it seems to me that many models relevant to agriPV will fit naturally into existing modules.  For example, PAR models could go in `pvlib.spectrum` (PAR is basically just spectral mismatch for plants, right?) or `pvlib.irradiance`, and several relevant view factor calculations are already available in `pvlib.bifacial.utils`.

AdamRJensen, 2024-05-24:
> For example, PAR models could go in `pvlib.spectrum` (PAR is basically just spectral mismatch for plants, right?) or `pvlib.irradiance`, and several relevant view factor calculations are already available in `pvlib.bifacial.utils`.

I agree that it seems to make sense to put PAR functions in `pvlib.spectrum`.

echedey-ls, 2024-05-24:
I've kept the documentation API section on agrivoltaics. Should I remove it too?

Just a minor organization idea, the function can be under `pvlib.spectrum` namespace but in another file, currently I've put it inside `mismatch.py`.

AdamRJensen, 2024-05-24:
> I've kept the documentation API section on agrivoltaics. Should I remove it too?

Hmm I'd lean towards keeping it with it's module. 

> Just a minor organization idea, the function can be under `pvlib.spectrum` namespace but in another file, currently I've put it inside `mismatch.py`. 

What would the argument be for thar?

echedey-ls, 2024-05-24:
> Hmm I'd lean towards keeping it with it's module.

Okay, I don't have a strong opinion on that either.

> What would the argument be for thar?

I kinda dislike having too many functions of different topics in the same file. Not a problem to have it there for me; in fact, isolating one function in a file is kinda weird too, so I'm happy with it in `mismatch.py`. It is just another idea to take into account in case of a big-reorg.

cwhanse, 2024-05-24:
@echedey-ls @AdamRJensen I commented on echedey's GSoC issue. I think the PAR function should live in `pvlib.irradiance`.

wholmgren, 2024-05-25:
> Just a minor organization idea, the function can be under pvlib.spectrum namespace but in another file

This is my preference if we anticipate additional spectral models for agrivoltaics. I don't know much about it but anticipate that there may be interest in, say, integration of quantum efficiency curves and that naturally should be near the similar code for traditional PV. 

I'm not too enthusiastic about irradiance.py... this function only appears to potentially fit there because that module is already so broad in scope. (Recall previous discussion of breaking it up into transposition.py and other modules)

cwhanse, 2024-05-25:
But the new function in #2048 isn't a spectral model in the scope of `pvlib.spectrum`s code module description. It's just translating broadband DHI fraction from diffuse fraction for a restricted range of wavelength. I think it is another version of an irradiance separation model.

cwhanse, 2024-05-25:
We have three comment threads on this naming/organization: here, #2064 and #2048.

PR #2048 (2024-06-20): Agrivoltaics - PAR diffuse fraction model
 - [x] Closes #2047 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue wi [... cut]
closed 2024-06-20 by AdamRJensen