pvlib/pvlib-python #2049: Allow arbitrary IAM function in pvlib.iam.marion_diffuse

markcampanelli, 2024-05-12:
**Is your feature request related to a problem? Please describe.**
I am working on an application using `pvlib.iam.marion_diffuse`, and I would like to inject my own IAM function (from interpolated IEC 61853-2 data) as a parameter to `pvlib.iam.marion_diffuse`.

**Describe the solution you'd like**
Add the option for passing a callable to `pvlib.iam.marion_diffuse` instead of the existing strings for the builtin IAM model functions. (The existing strings would still be accepted.)

Also add a function to create IAM functions as a PCHIP interpolation of data such as from IEC 61853-2. Note that this solution can also be made to add support for passing the existing `pvlib.iam.interp` function (whose `cubic` option is slightly different that `pchip`) to `pvlib.iam.marion_diffuse`.

**Describe alternatives you've considered**
Rolling my own interpolant each time, and then calling the lower-level `pvlib.iam.marion_integrate` for each of the three components.

**Additional context**
This is a followup to https://github.com/pvlib/pvlib-python/issues/1980, but without any concern to improving the diffuse-horizontal integral.

[image]

adriesse, 2024-05-13:
Have you considered adding the pchip option into ```pvlib.iam.interp``` rather than creating a new function?

markcampanelli, 2024-05-13:
@adriesse 

> Have you considered adding the pchip option into `pvlib.iam.interp` rather than creating a new function?

I am open to this consolidation. Do you think folks would be open to `pvlib.iam.interp` returning a `callable` that could then be passed to the new `pvlib.iam.marion_diffuse` (or used for other calculations)?

If I understand correctly, this avoids some overhead of recreating the interpolating function on each usage, although it would break the existing interface to `pvlib.iam.interp`.

adriesse, 2024-05-14:
> @adriesse
>
>
> I am open to this consolidation. Do you think folks would be open to `pvlib.iam.interp` returning a `callable` that could then be passed to the new `pvlib.iam.marion_diffuse` (or used [...]

I think Marion only has one call to the model.

echedey-ls, 2024-07-01:
Feel free to break down these issues into different PRs, doing too many things in one big PR makes the review process more time consuming - and reviewers more hesitant to review.

closed 2026-11-14 by markcampanelli