pvlib/pvlib-python #2057: Support for HJT modules

fredrbl, 2024-05-22:
As far as I can see, the `pvlib.ivtools.sdm.fit_cec_sam` function does not accept HJT modules. In the time coming, more and more HJT type modules (e.g HIT) will be used, so it would be nice if pvlib could support it.

Are there any other ways of achieving the CEC single diode model parameters for HJT modules? Or, in general any way of running the single diode model with HJT modules in pvlib?

Context:
Our starting point is PAN files. I cannot find any way of converting HJT modules from PAN files to correct single diode model parameters.

kandersolar, 2024-05-22:
This thread on the google group discusses the same question (what cell_type to use for HJT in fit_cec_sam): https://groups.google.com/g/pvlib-python/c/6AdD14aSPlc/m/Dbx8B--kAwAJ

A related question: if you have a PAN file, why not use the PVsyst SDM instead of the CEC SDM?

cwhanse, 2024-05-22:
`fit_cec_sam` is merely a convenience to use to CEC fitting tool in SAM. I think SAM is fitting HIT/HJT modules as if they were mono-c-Si, see the entry for "SANYO ELECTRIC CO LTD OF PANASONIC GROUP HIT-N210A01". You can try getting CEC coefficients that way.

And as a teaser, a colleague here at Sandia will present on this subject (converting between single diode models) at PVSC in two weeks.

markcampanelli, 2024-06-24:
@fredrbl I did a fitting intercomparison with [9 modules](https://pvpmc.sandia.gov/datasets/pv-lifetime-module-datasets-clone/) recently, and the one HIT module (from Panasonic) was a distinct outlier in quality of fit to the particular SDM that I used. In those fits I allowed the band gap to vary as a fitting parameter (which I _think_ is what the `Technol` parameter basically controls in PVsyst), and even this flexibility did not produce nearly as good of fits as for the 8 x-Si modules. I think examining SDM fits for HJT technologies is worth a closer look at this point.

markcampanelli, 2024-06-28:
@mikofski I always forget that SolarFarmer arguably has better documentation of PVsyst’s model than PVsyst 🤣.

cwhanse, 2024-06-28:
@fredrbl I'm not sure we've addressed your concern. Can you let us know if the discussion has helped clarify and perhaps close the issue?

fredrbl, 2024-07-01:
Hello @cwhanse, sorry for the delay here.
Sort of! On the one hand, it seemed OK to set Technol to Monocrystalline for the HJT cases, but then I got worried when I saw @markcampanelli comment about the outlier. So I am still unsure how "off" the results might be with the assumption.

And! @kandersolar , what function do you mean when PVsyst SDM? Are you referring to `pvlib.ivtools.sdm.fit_pvsyst_sandia`?
From this comment: `A related question: if you have a PAN file, why not use the PVsyst SDM instead of the CEC SDM?`.
If so, I don't have a good reply; its just the way we do our simulations (we don't always have the PAN file for panels, some times we just have to type in the necessary parameters from the data sheet)

kandersolar, 2024-07-01:
I meant `pvlib.pvsystem.calcparams_pvsyst`, combined with `pvlib.pvsystem.singlediode`.  Sticking with the PVsyst model, there's no (technical) need to call some kind of `fit_` function, since the PAN file already supplies the necessary parameters.  But if a PAN file is not always available, and you want to use an approach that works in all cases, then the process you describe seems reasonable to me.

kandersolar, 2024-10-19:
Closing as it seems like the question is answered.  @fredrbl feel free to reopen if not, although questions like this are better posted to the pvlib google group :)

closed 2024-10-19 by kandersolar