pvlib/pvlib-python #2064: GSoC participation roadmap of Echedey

echedey-ls, 2024-05-23:
Hi everybody!

After a welcoming meeting I had with this year GSoC mentors, @kandersolar and @AdamRJensen, and the other two participants, @RDaxini and @IoannisSifnaios, we were recommended to have a tracking issue of the milestones for each one of us. Here's mine.

# Roadmap
- Martinez shading losses model
  -
  - Depends on merging #1962 
  - #2063
  - No PR open yet
- Spectra analysis
  -
  - Extend standard spectra database
    - #1963 
    - #2039 
  - Add Spectral response <> Quantum Efficiency conversion functions
    - #2040
    - #2041
  - Add spectral responses of materials
    - #2037 
    - #2038
- Non-uniform irradiance mismatch losses model
  -
  - #1541 
  - #2046
- Photosynthetically Active Radiation models
  -
  - Decomposition: fraction of diffuse (Spitters relationship)
    - #2047 
    - #2048
  - Diffuse and direct shading factors - 3D geometry calculation of shades via geometry and integration.
    I'm afraid this one is doing PVsyst-like calculations all over the 3D and 2D space, so it's gonna take some time to sort out all the difficulties and integrate it here in pvlib.
    Here is [the paper](https://www.sciencedirect.com/science/article/pii/S0306261923003458) meanwhile.
    - No issue yet
    - No PR yet

I want to open issues and PRs as I work on them. If I can manage so many things at the same time is something I have to figure out, not you - so don't hesitate to give as much feedback as you want! 😆 

I'm looking forward to use my knowledge to improve this package, but as the inexperienced student I am, I can't do it without y'all, so I want to hear from you in either issues or PRs 🤙 

Cross-referencing:
- [My GSoC mailing thread](https://groups.google.com/g/pvlib-python/c/TL455jQWKco/m/CLrfZHyYBAAJ)
- [Personal blog](https://echedey-ls.github.io/)

cwhanse, 2024-05-24:
Thinking about the API for the agrivoltaics functions, modeling an agrivoltaic system involves a modeling chain that handles both the photovoltaic and crop production. Some steps in that chain are in common, e.g., separating GHI into diffuse and direct, but more likely, many steps are in parallel, e.g. calculate cell temperature and calculate air temperature for the crops. The same function won't be used for both steps.

Two options suggest themselves:

1. put agrivoltaics functions into the code module for the corresponding photovoltaic modeling step, i.e., irradiance, temperature, unless there's no corresponding photovoltaic step, in which case the function goes into an `agrivoltaic` module, e.g., interface function to crop production model libraries.
2. put all agrivoltaics functions into a separate module(s) the scope of which is only agrivoltaics.

I lean to #1, since I prefer to look at agrivoltaics as a variation of the core pvlib workflow. I have the same view of floating PV modeling.

If it's #1, then the PAR diffuse fraction function would go with the rest of the GHI separation functions, into `pvlib.irradiance`.

echedey-ls, 2024-09-06:
This GSoC has come to its end, so this issue can be safely closed :D 

❣️ Thank you very much to everybody, specially the mentors @kandersolar & @AdamRJensen and fellow contributors @RDaxini & @IoannisSifnaios for the funny meetings, and all the maintainers and community members involved in my PRs and discussions (@cwhanse , @adriesse , @mikofski ...)

I will try to continue contributing to pvlib in the future as well 🫡

closed 2024-09-06 by echedey-ls