pvlib/pvlib-python #2068: Google Summer of Code: Floating PV

IoannisSifnaios, 2024-05-26:
This issue gives an overview of the plan and progress of my [GSoC ](https://summerofcode.withgoogle.com/) project.

The title of my project is "Floating Solar - Enabling modeling of floating PV with pvlib", where I will be adding floating PV temperature models in the pvlib python’s library. The official project description can be found [here](https://summerofcode.withgoogle.com/programs/2024/projects/Z32UN1Qx), and my mentors during this project will be @AdamRJensen and @kandersolar. My detailed plan is the following:

## Calculation of ambient parameters for floating PV:

- Albedo for water bodies (non-ocean) as described [here](https://ttu-ir.tdl.org/server/api/core/bitstreams/7a06ecbe-c254-44c7-9ee2-d49dbb3bd9f3/content)
- Is relevant to #1227 

## Module temperature for floating PV:

- Gallery example of the expansion of the Faiman expression for floating PV (using the methodologies from [here ](https://www.mdpi.com/1996-1073/16/20/7153)and [here](https://www.sciencedirect.com/science/article/pii/S0038092X21002085))

- Addition of the Lindholm model (described [here](https://onlinelibrary.wiley.com/doi/abs/10.1002/pip.3451))

- Addition of the Rahaman thermal model (described [here](https://www.sciencedirect.com/science/article/pii/S0960148123003713))

As the project develops, I plan to link these tasks to individual issues and pull requests.

---

Please don't hesitate to give any feedback! Since this is the first time I am contributing to pvlib (or any other open-source software, tbh), I am really looking forward to learning as much as possible from all of you!

IoannisSifnaios, 2024-05-26:
It looks like there is already a plan to add a `floating.py` module (as mentioned in #1937).

IoannisSifnaios, 2024-08-08:
My GSoC is completed, thus this issue is closed. Thanks to everyone for their assistance!

kandersolar, 2024-08-08:
Thanks @IoannisSifnaios for the great contributions!