pvlib/pvlib-python #2087: add JRC spectral factor model

RDaxini, 2024-06-15:
**Is your feature request related to a problem? Please describe.**
The PVSPEC model for the spectral mismatch factor was recently merged (#2072) but an alternative parameterisation of the mismatch factor as a function of air mass and clear sky index is available, namely [the JRC model](https://www.researchgate.net/publication/256080247_A_simple_model_for_estimate_the_influence_of_spectrum_variations_on_PV_performance). I think providing pvlib users with a choice of model would be a valuable enhancement.

**Describe the solution you'd like**
Implement JRC model using the PVSPEC function as a template.

**Additional context**
Some questions:

1. Citation for the clear sky model in [the JRC publication](https://www.researchgate.net/publication/256080247_A_simple_model_for_estimate_the_influence_of_spectrum_variations_on_PV_performance) is:
>"K. Scharmer and J. Greif (eds.), “The European Solar Radiation Atlas, Vol 2: Database and Exploitation Software”, Presses de l’École des Mines, Paris (2000)"

and

>"M. Šúri and J. Hofierka. “A new GIS-based solar radiation model and its application to photovoltaic assessments”, Transactions in GIS, 8 (2004), pp. 175-190".

In [the PVSPEC paper](https://ieeexplore.ieee.org/document/9300932) and [the pvlib implementation](https://pvlib-python--2072.org.readthedocs.build/en/2072/reference/generated/pvlib.spectrum.spectral_factor_pvspec.html):
> "Rigollier, C., Bauer, O. and Wald, L., 2000. On the clear sky model of the ESRA—European Solar Radiation Atlas—with respect to the Heliosat method. Solar Energy, 68(1), pp.33-48.". 

I think these are the same ESRA model (?) ---if so, which citation should I use?

2. The model used to estimate air mass is not stated, nor whether it is relative or absolute air mass. Doesn't affect the maths, but in terms of docs is it sufficient to write something like "the model used to estimate the air mass (denoted AM) is not stated." and leave it at that?
----
This issue links to my GSoC project (#2065) and this issue #1950

echedey-ls, 2024-06-15:
> As far as I can tell, the model used to estimate air mass is not stated, nor whether it is relative or absolute air mass.

The measurements facility is roughly 220m above sea level, so the difference in absolute and relative AM may not be that significant? Talking from my unknowledgeable point of view. We could maybe ask the authors?

echedey-ls, 2024-06-20:
I've seen you've changed from an unqualified airmass to `airmass_absolute` in the PR. Have you got into contact with the authors, have you found that information anywhere or did I convince you? My guess intentions just were to speed up some initial work.

RDaxini, 2024-06-20:
@echedey-ls I have already reached out but no response yet. I will leave it as airmass as that is what is written in the paper. No reason to assume otherwise. Citations of the manuscript do the same, e.g. https://doi.org/10.1016/j.solener.2013.05.011

PR #2088 (2024-06-20): Add JRC spectral factor correction

 - [x] Closes #2087; partially closes #2065 and #1950 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with [... cut]
closed 2024-06-20 by AdamRJensen