pvlib/pvlib-python #2089: Inexistent function reference in `pvlib.irradiance.poa_components`

echedey-ls, 2024-06-15:
The following referenced function does not exist:
https://github.com/pvlib/pvlib-python/blob/19c9598355b8243038bdeb1253d219e0371f71af/pvlib/irradiance.py#L474

A alternative should be found and referenced with ```:py:func`pvlib.<submodule>.<function_name>` ```

cwhanse, 2024-06-15:
Should point to :py:func`pvlib.irradiance.get_ground_diffuse`

echedey-ls, 2024-06-15:
Good first issue for newcomers or do you prefer a PR?

AdamRJensen, 2024-06-15:
> Good first issue for newcomers or do you prefer a PR?

I like the idea of leaving it, so we have something in case someone comes by.

studentayaaljabbari, 2025-06-16:
I'd like to work on this issue. From what I understand, we need to:
1. Fix the documentation reference in `pvlib/irradiance.py` line 474
2. Change the reference from the non-existent `grounddiffuse` function to `pvlib.irradiance.get_ground_diffuse`
3. Use the correct Sphinx documentation format: `:py:func:pvlib.irradiance.get_ground_diffuse`