pvlib/pvlib-python #2091: question about pvefficiency_adr

Typezero-shanxi, 2024-06-17:
I've come across another thing I don't understand when running the official case, and would like an answer to see if my understanding is wrong.
[image]
It is clear that the relative efficiency will be greater than 1 as shown on the graph, what to make of this situation. Also, the code in the program is as follows:
df['p_mp'] = P_STC * df['eta_rel'] * (df['poa_global'] / G_STC)

The choice of values for G_STC and P_STC can significantly affect the output of df['p_mp'], so is there any basis for the choice of the two?

Many thanks for the attention and answer ！！！！！

AdamRJensen, 2024-06-17:
@Typezero-shanxi Please post user questions in the pvlib Google Group or start a [discussion](https://github.com/pvlib/pvlib-python/discussions).

The GitHub issues are for reporting bugs and development proposals.

closed 2024-06-17 by AdamRJensen

Typezero-shanxi, 2024-06-17:
Thanks for your answer