pvlib/pvlib-python #2096: Document internal module-level constants

echedey-ls, 2024-06-17:
**Is your feature request related to a problem? Please describe.**
A lot of times constants are referenced in the documentation, but the only way to access them is either by visual inspection of the code or importing the object that contains it programmatically.

**Describe the solution you'd like**
Include constants in a toctree and add their respective documentation.

**Describe alternatives you've considered**
A toctree dedicated exclusively to all pvlib constants.

**Additional context**
https://stackoverflow.com/questions/40748886/how-can-i-document-a-constant-module-level-variable-with-sphinx-docstring-wit
https://stackoverflow.com/questions/50831070/how-to-document-linkable-constants-with-sphinx

kandersolar, 2024-06-18:
To my knowledge, the current sphinx docs document only one module-level variable; see the bottom of this page: https://pvlib-python.readthedocs.io/en/stable/reference/pv_modeling/temperature.html

I think documenting the variables near the functions they are related to makes sense.  I don't think a toctree for all variables makes sense.

mikofski, 2024-06-18:
Should we consider properties for constants. Although highly unlikely, constants could be accidentally overwritten and currently nothing in pvlib would alert you to this. Although clumsy a property is read only and cannot be changed without a `set` method

echedey-ls, 2024-06-18:
> Although highly unlikely, constants could be accidentally overwritten and currently nothing in pvlib would alert you to this

I'm only in favour about marking their type as [Final](https://docs.python.org/3/library/typing.html#typing.Final). It would alert linters.
The same way we almost never check input values to models because we expect the user-base to know what they are doing, I would not try to over-complicate providing some handy values.

mikofski, 2024-06-19:
Agreed, I only mention it here for posterity in case this becomes an issue again in the future. The documentation of ` TEMPERATURE` was previously an issue, & I had suggested [making it a class](https://github.com/mikofski/pvlib-python/blob/temp_model_params/pvlib/temperature.py) which allows it & its members to be documented. This was a drop in replacement, although admittedly unnecessarily overcomplicated

anshurajbisoyi98-ctrl, 2026-10-04:
Hi, I’d like to contribute to this issue. Would a focused PR documenting `SURFACE_ALBEDOS` alongside the related irradiance functions, including its source reference, be a useful starting point?

cwhanse, 2026-10-04:
@anshurajbisoyi98-ctrl that would be welcome. Please ensure any AI content is vetted before you submit.