pvlib/pvlib-python #2099: Suggestion to change the default solar constant in get_extra_radiation()

kurt-rhee, 2024-06-20:
**Is your feature request related to a problem? Please describe.**
The default solar constant in pvlib's `get_extra_radiation` function is 1366.1.  This default contradicts with the recommended solar constant in the default model's reference (Spencer).

 * **pvlib**:  https://pvlib-python.readthedocs.io/en/stable/_modules/pvlib/irradiance.html#get_extra_radiation
 * **Spencer Reference**:  http://solardata.uoregon.edu/SolarRadiationBasics.html
 * **Another Reference**: https://agupubs.onlinelibrary.wiley.com/doi/full/10.1029/2010GL045777

**Describe the solution you'd like**
I would like to propose changing the default to 1360.8 per the references provided above.

**Describe alternatives you've considered**
* We could leave the current value as the default and provide some information in the docstring as to why the default value and the values assumed in the reference materials are different.

echedey-ls, 2024-06-20:
We could also make the solar constant optional, then default to the one recommended by each model? A bit of over-engineering for such small differences, but possibly the best approach to be as accurate as possible with the references.

adriesse, 2024-06-21:
Many (all?) models were developed with specific values, so it's probably best if they default to those.

kurt-rhee, 2024-06-21:
Agreed with @echedey-ls  and @adriesse that each model should default to the values that they were developed with. 

I am not familiar with many of the models, but I will give them a read through and make suggestions when I have some free time and am back from travels:

- spencer: 1361.0
- asce: ?
- pyephem: ?
- nrel: ?

kurt-rhee, 2024-06-21:
I would also like to propose that we note which reference corresponds with which model choice.  Does anybody know if there are similar patterns elsewhere in the project that I can copy?

cwhanse, 2024-06-21:
Before we get too far, which modeling steps involve the solar constant? And, is it a problem if different values are used at each step?

kurt-rhee, 2024-06-21:
I wouldn't say that I am an expert on all of the modeling steps, but my understanding is that `solar_constant` only gets used to calculate the extraterrestrial DNI and then the models further down the model chain use extraterrestrial DNI as an input.

A quick search of pvlib though shows that there are two places where the term `solar_constant` is used:
1. In the calculation of extraterrestrial DNI as expected.
2. In the calculation of bird simple spectral model

Reference for 2:
https://pvlib-python.readthedocs.io/en/stable/_modules/pvlib/irradiance.html?highlight=solar_constant

```
    # ET spectral irradiance correction for earth-sun distance seasonality.
    # Note that we only want the distance correction coefficient, so set
    # solar_constant=1:
    earth_sun_distance_correction = \
        pvlib.irradiance.get_extra_radiation(dayofyear, method='spencer',
                                             solar_constant=1)  # Eq 2-2, 2-3
```

My thinking is that since, we are proposing only to change the default by model, these explicit over-rides will still be available for people to use.

echedey-ls, 2024-06-24:
Ref. [4] states:

> A value of Gsc of 1367 W/m2 (1.960 cal/cm2 min, 433 Btu/ft2 h, or 4.921 MJ/m2 h) is used in this book.

https://breadl.org/d/275235 use mirror link no. 1 (IDK how much uptime do this pages have)

adriesse, 2024-06-25:
It would make a nice student project to research this topic more deeply present a summary report.

echedey-ls, 2024-07-07:
I have just seen this issue is a duplicate of #1566