pvlib/pvlib-python #2105: Unused fixture input in a fixture

echedey-ls, 2024-06-22:
Was the following piece of code supposed to use it's input?

https://github.com/pvlib/pvlib-python/blob/55e0a36fac1e3931b92f5f11508f6c09d903e0cc/pvlib/tests/test_irradiance.py#L61C1-L65C30

My suggestion is to remove the input parameter; I've run the tests without it and that works locally.

I suppose it's not a big deal, and for the same reason, another nice good first contribution.

wholmgren, 2024-06-22:
I think it's a holdover from when `dni_et` (and other inputs) were calculated inputs rather than fixed inputs to various tests. It was easy to get started with calculated inputs but we eventually realized that led to unhealthy coupling in the test suite. So, safe to remove the unused fixture there.

Clemo97, 2024-07-17:
Was this issue fixed??

echedey-ls, 2024-07-17:
Hi @Clemo97 !

No, it hasn't. Would you like to work on this?

I recommend reading the how to contribute page https://pvlib-python.readthedocs.io/en/stable/index.html

And this amazing post from @kurt-rhee https://kurt-rhee.github.io/2024/06/13/how-to-contribute-to-pvlib

We would be happy to help you get involved :)

PR #2156 (2024-08-07): Removed unused input as shown in issue #2105
 - [x] Closes #2105 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] Pull request  [... cut]
closed 2024-08-07 by kandersolar