pvlib/pvlib-python #2107: add spectrum.spectral_factor example

RDaxini, 2024-06-24:
**Is your feature request related to a problem? Please describe.**
There is currently no example in the example gallery demonstrating the use/application of the `spectrum.spectral_factor` models.

**Describe the solution you'd like**
I am currently working on an example that demonstrates the calculation of spectral mismatch using multiple spectral factor functions.

**Additional context**
Related issue: #2065 
I am away for a short while so won't have the PR ready this week, so in the meantime I wanted to open this issue in case anyone has any feedback or thoughts about anything in particular they'd like to see in such an example that they feel would be beneficial.

I am still deciding a few things, such as analysis duration and data source (currently thinking one day and NREL MIDC --- any other suggestions welcome) but will create a PR once I have something concrete.

echedey-ls, 2024-06-24:
> data source 

Spectral data can also be created synthetically with a spectral model, like [`pvlib.spectral.spectrl2`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.spectrl2.html). May be out of scope for an example, thou.

RDaxini, 2024-06-24:
@echedey-ls thanks for your input
The example will use spectral mismatch models currently in pvlib, which don't require spectral irradiance data.
The stretch goal of the GSoC project is to implement a spectral factor model that requires spectral irradiance data, in which case synthesised spectral irradiance data may be useful if the example were to be updated to include the new model

echedey-ls, 2024-06-24:
That's perfect. I thought you wanted to compare the spectral mismatch of some model with the specific mismatch of a technology.

kandersolar, 2024-06-28:
> I am still deciding a few things, such as analysis duration and data source (currently thinking one day and NREL MIDC --- any other suggestions welcome)

If the data source is to provide a "ground truth" SMM value, I'd say no need.  It is enough to demonstrate the model's use; validating the models against real data is out of scope IMHO.

If the data source is for model inputs, I'd prefer something like the NSRDB over ground station data.  It could be that pvlib already has a PSM3 or TMY file with the necessary inputs in `pvlib/data` (although be warned, it's a disorganized in there).

RDaxini, 2024-06-30:
@kandersolar The purpose is for model inputs rather than providing a "ground truth" SMM value. Thanks for the recommendation, I'll take a look!