pvlib/pvlib-python #2113: Should marion_diffuse use diffuse versions of martin_ruiz and schlick?

markcampanelli, 2024-06-30:
**Describe the bug**

While working on https://github.com/pvlib/pvlib-python/pull/2050, it occurred to me that `pvlib.iam.marion_diffuse` refers to the direct component IAM functions for `martin_ruiz` and `schlick`, i.e., it would seem that [these two lines](https://github.com/pvlib/pvlib-python/blob/main/pvlib/iam.py#L630-L631) should instead refer to `pvlib.iam.martin_ruiz_diffuse` and `pvlib.iam.schlick_diffuse`

**To Reproduce**
Steps to reproduce the behavior:

1. Inspect the lines at: https://github.com/pvlib/pvlib-python/blob/main/pvlib/iam.py#L630-L631

**Expected behavior**
I would expect the diffuse IAM models to be used, because marion_diffuse computes diffuse components.

**Screenshots**
None

**Versions:**
 - ``pvlib.__version__``: v0.11,0
 - python: 3.11.7

**Additional context**
If someone could verify, then I would be happy to fix this as part of https://github.com/pvlib/pvlib-python/pull/2050.

echedey-ls, 2024-06-30:
If I understand correctly, that diffuse IAM uses direct IAM models to compute a diffuse one via integration. So it sounds good to me. It uses this function: https://github.com/pvlib/pvlib-python/blob/13ef8f4e3f07a6b1c5acd7014a2617beeb10867c/pvlib/iam.py#L647C1-L653C34

markcampanelli, 2024-06-30:
> If I understand correctly, that diffuse IAM uses direct IAM models to compute a diffuse one via integration. So it sounds good to me. It uses this function: https://github.com/pvlib/pvlib-python/blob/13ef8f4e3f07a6b1c5acd7014a2617beeb10867c/pvlib/iam.py#L647C1-L653C34

I think I understand now: `pvlib.iam.martin_ruiz_diffuse` and `pvlib.iam.schlick_diffuse` are analogous to `pvlib.iam.marion_diffuse`, but with a slightly different output interface. I was wondering why one would need a separate IAM function for direct vs. diffuse 🤪. Thanks @echedey-ls!

markcampanelli, 2024-06-30:
Closing due to misunderstanding of code.

markcampanelli, 2024-06-30:
Closing due to misunderstanding of code. (Github's issue-closing interface is a bit wierd!)

markcampanelli, 2024-06-30:
Really, close this.

markcampanelli, 2024-06-30:
Really, close this.

closed 2024-06-30 by markcampanelli