pvlib/pvlib-python #2117: Add  Hofierka 2002 model

AdamRJensen, 2024-07-05:
This model is currently being used by PVGIS but is not available through pvlib at the moment.

kandersolar, 2024-07-05:
I think the full reference is Jaroslav Hofierka, Marcel Suri, The solar radiation model for Open source GIS: Implementation and applications.  Proceedings of the Open source GIS - GRASS users conference, 2002.

Available here: https://www.researchgate.net/publication/2539232_The_solar_radiation_model_for_Open_source_GIS_Implementation_and_applications

AdamRJensen, 2024-07-05:
Most of the modeling steps described in this publication are no longer being used by PVGIS as they get GHI/DNI from external sources (satellite derived).

I think the really only useful step to add to pvlib in order to be able to replicate pvgis is the Muneer transposition model:

Muneer, T., 1990, Solar radiation model for Europe. Building services engineering research and technology, 11: 153-163.