pvlib/pvlib-python #2118: Expression for calculating wind speed at different heights

IoannisSifnaios, 2024-07-05:
Most times, the wind speed provided in weather files is measured at a height of 10 m. However, in many cases it is required to estimate the wind speed at panel height. Which equation should be used in order to do this?

I found this issue (#1814), which mentions an equation from a Sandia report (I couldn't find the report, though... some help from Sandia people?).

And then there is the expression used in wind engineering, described [here](https://en.wikipedia.org/wiki/Wind_gradient) (for a more official reference, it is also mentioned in the book by Siegfried Heier (2014), Grid Integration of Wind Energy Conversion Systems (publisher: Wiley))

I am willing to take up the task of making the PR, etc., but first, I would like to know your thoughts on which expression should be added (maybe both?) or if there is some other equation you think is more relevant.

kandersolar, 2024-07-05:
Here is a public copy of SAND 84-2530: https://web.archive.org/web/20230418202422/https://www2.jpl.nasa.gov/adv_tech/photovol/2016CTR/SNL%20-%20Est%20Wind%20Speed%20vs%20Height_1985.pdf

cwhanse, 2024-07-05:
A word of caution here: some temperature model coefficients (e.g. those for the SAPM) implicitly account for the difference between anemometer and PV module height above ground. They do so by regressing measured module temperature to wind speed at 10m.

AdamRJensen, 2024-07-05:
> A word of caution here: some temperature model coefficients (e.g. those for the SAPM) implicitly account for the difference between anemometer and PV module height above ground. They do so by regressing measured module temperature to wind speed at 10m.

Great insight @cwhanse. Let's add this info in a warning admonition for the proposed wind speed function.

Probably worth checking out if there is similar functionality in metpy.

IoannisSifnaios, 2024-07-05:
Good point, @cwhanse! From a quick search, I can see that only the SAPM models and the generic_linear require the wind speed of 10 m as input. The rest of the temperature models either require module height or don't specify (e.g., Fuentes), but I would still imagine it requires module height...

mikofski, 2024-07-05:
I agree with @cwhanse we we need to tread cautiously because in my experience there is already a lot of misunderstanding about correlation coefficients and their relation to wind speed height measurements.

1. I emphasize what Cliff says: correlation coefficients like `Uc = 25 W/m2/K` and `Uv = 1.2 W/m2/K/(m/s)` rewrite and depend on wind speed measured at 10-m. So if wind speeds were measured at array height, eg 2-m, a function such as the on proposed could be helpful in estimating wind speed at 10-m
2. Ideally I would prefer to see a discussion added to the documentation explaining why free stream velocity measurements are preferred and used in temperature correlations, before adding new functions
3. I have most frequently seen a power law but there is often debate over what exponent to use. Eg: `Uw(10m) = Uw(2m)*(10m/2m)^n` where `n` is the exponent. Other methods may be more accurate but I’m not familiar with them. I also don’t have a reference for this or know the history why it’s so prevalent

adriesse, 2024-07-05:
This is my view on the subject:

> B. Wind speed
>
>Each model has a coefficient that multiplies wind speed. In practice, wind speed increases with height above the ground in a non-linear manner, therefore the model coefficients are valid for a specific combination of module installation height and wind speed measurement height. The Faiman model coefficients and NOCT values are typically determined using wind speed measured near module height, but the modules are not necessarily near ground level. SAPM and PVsyst coefficients are specified for use with wind speed data at the standard 10 m height, but that information is only useful if the module height corresponding to those coefficients is also clearly specified.
>
>To accommodate a different installation or measurement height, either the wind speed data or the wind speed coefficient can be adjusted. The basis for such an adjustment is often either the log law or the power law which describe the wind speed profile as a function of height using an empirical parameter related to the surface roughness; **however, these profiles are not applicable close to the ground or close to the level of the objects that contribute to the roughness.** [12] For the adjustment from 10 m to 2 m, a nominal height for a ground-mounted array, some reduction ratios found in the literature are: 0.51 [7], 0.56 [13], 0.67 [14] and 0.725 [15]. This range gives an indication of the level of uncertainty associated with such adjustments.
>
>The SAM NOCT model uses a predetermined factor of 0.51 to reduce 10 m wind speed data to ground-mounted, module-level wind speed, but unfortunately the origin of this value is undocumented. The need for adjustments with other models to or from different heights should be evaluated by the user of the models on a case by case basis taking into consideration the both the source of the empirical parameters and the source of the wind speed data.

mikofski, 2024-07-06:
Thanks Anton! Can we get those references?

adriesse, 2024-07-06:
Here is a link to my temperature model equivalence paper that contains the above excerpt: 
https://www.osti.gov/biblio/2003640

mikofski, 2024-07-06:
Thanks Anton. So I think there is a case for pvlib to include an expression to account for wind speed height measurement, but IMO. I still think the documentation takes precedence so that folks don’t misuse it. In particular I’d like to see a table of weather resources that shows the wind speed height. Eg ground campaigns most often measure at about 3-m while satellite is usually at 10-m. Also a corresponding table of temperature models. Finally some gallery examples that show when and how an adjustment should be made. My personal preference is to adjust the wind coefficient rather than the wind speed measurements which is associative (ie: `Uv’ * Uw = Uv * adj * Uw = Uv * Uw’` in which prime indicates adjustment. Also a discussion of uncertainty would be useful.

adriesse, 2024-07-08:
I think this is another area where the PV modellers are just muddling along.  Someone needs to access a large amount of data and do some thorough analysis in order to develop some well-founded correlations that are useful for PV systems!

adriesse, 2024-07-24:
Do we need consensus here before the related PR is merged?

cwhanse, 2024-07-24:
What concerns are outstanding? I'm going back through the comments and it would help to summarize.

adriesse, 2024-07-26:
Let's go back to the beginning:

> Most times, the wind speed provided in weather files is measured at a height of 10 m. However, in many cases it is required to estimate the wind speed at panel height. Which equation should be used in order to do this?

Two assertions and a question. 

The answer to the question is: Not the power law.  Especially not if there we're talking about the height of an actual row or field of panels at that location rather than the panel height in an empty field.  

The second assertion invites the question:  Does it really happen often, that wind speed at panel height is needed?  For me, this is not the case, but perhaps for others it is.

kandersolar, 2024-07-30:
Maybe (probably?) the simple power law is poorly suited to translating wind speeds in PV modeling contexts, but I don't think there is any doubt that it nevertheless sees wide use in that application.  I just noticed it in IEC 61724-2:2016 Appendix A, for example.  

My two cents: widespread usage is sufficient motivation for inclusion in pvlib, despite theoretical concerns with the model.

adriesse, 2024-08-01:
> Maybe (probably?) the simple power law is poorly suited to translating wind speeds in PV modeling contexts, but I don't think there is any doubt that it nevertheless sees wide use in that application. 

I have some doubts about the widespread use just like I have doubts about the frequency of the need. 

> I just noticed it in IEC 61724-2:2016 Appendix A, for example.

This is an informative appendix rather than normative, and it is particularly carefully worded to avoid any recommendation.

> My two cents: widespread usage is sufficient motivation for inclusion in pvlib, despite theoretical concerns with the model.

This is a good example to motivate debate about more general policies for inclusion in pvlib.  I would argue that widespread usage is never enough by itself, and it should not trump scientific concerns.

kandersolar, 2024-08-01:
I see plenty of use of it here: https://scholar.google.com/scholar?hl=en&as_sdt=0%2C32&q=%22photovoltaic%22+%22power+law%22+temperature+height+10+meter+-hub&btnG=

Frequency of need: anemometers installed at solar farms are at roughly module height, not the 10 m assumed by thermal models.  The NSRDB reports wind speed at 2 m, if I remember correctly.

I'll leave discussion of the last point for another day.  For now, I think there are enough thumbs up here to proceed with #2124.

adriesse, 2024-08-04:
> For now, I think there are enough thumbs up here to proceed with #2124.

I don't find that the foregoing discussion justifies this conclusion.  

But, I have thought of one potential way to turn the inclusion of this model in pvlib into a positive thing.  If this model can be coded and documented in such a way that it draws attention to and increases awareness of its limitations, shortcomings and uncertainties, and makes people think before they leap (or simulate), then it could indirectly make this world a better place.

We could furthermore consider whether including the log law at the same time could contribute even more to that goal, by forcing users to think even more about the modeling challenge.  This log model has an additional parameter for surface effects, although that doesn't mean it's been validated for near-surface wind speeds either (like in a field of PV modules).

And just maybe some big data hoarder will stumble on this and think: hey, I can validate this!

closed 2024-08-05 by AdamRJensen

wholmgren, 2024-08-05:
Great ideas @adriesse. Happy to work with you on follow up PRs to add the log law and improve the documentation. I don't like the fact that billions of dollars of PV have been financed using resource data (or coefficients) touched by the power law. I think the reference implementation documented with shortcomings will be useful as we try to prevent future mistakes.

adriesse, 2024-08-05:
@wholmgren You're on your own on the follow-up PR.  My concerns weren't taken seriously on this one, so I don't expect any better next time.

PR #2124 (2024-08-05): Function to estimate wind speed at different heights
 - [X] Closes #2118 
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [X] Tests added
 - [X] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitH [... cut]