pvlib/pvlib-python #2121: A ModelChain object to batch simulate PV for multiple different Locations and different PVsystems

BYC12138, 2024-07-08:
I found that instantiating a ModelChain can batch simulate PV for multiple different Locations and different PVsystems.
    My task is like this, I have meteorological data, radiation data and PV panel mounting angle data for a certain region at a certain moment in time, they are all originally a 2D array. I need to simulate the PV power generation in this region at this moment in time, following the tutorial on the pvlib website and the comments in the source code:
    Stretch this data into a table, with each row representing a Location, instantiate a PVsystem based on the angle of the PV panels installed at this location, which in turn instantiates a ModelChain, and enter the weather data from this row as weather into the ModelChain for simulation. Then need to traverse each row of data in this table to simulate, that is, each row of data need to instantiate a new ModelChain, because each row of my data represents a different Location, a different PV panel installation angle. And finally reassemble them into a 2D array.
    But my sis realized that it can be done this way:
    Stretch these data into a table, instantiate the whole columns (latitude,longitude) as a pvlib.location, and instantiate the whole columns of PV panel mounting angles (surface_tilt,surface_azimuth) as a PVsystem, which in turn instantiate a ModelChain finally The weather data of the whole table is input into ModelChain as weather for simulation. Thus, only one ModelChain needs to be instantiated to simulate the PV power generation at all locations. Finally, they are then reassembled into a two-dimensional array. Without traversing tens of thousands of rows of samples, the run is naturally ridiculously fast. And the results should be correct as well.
    I explored pvlib's modelchain.py source code in the spirit of scientific rigor, and I found that the implementations of the ModelChain model do support matrix operations, i.e., they can be batch computed (note that I'm not talking about the batch simulation of multiple pieces of weather data for a Location and a PVsystem): instantiated shape same Location, PVsystem, instantiate those Locations, PVsystems into a ModelChain, and simulate the weather data with the same shape once.
    Also, I found that the instantiation of whole columns (latitude,longitude) into a single pvlib.location on version 0.10.3 of pvlib is working fine, but on version 0.11.0 it reports an error (as in Figure 1). I explored the source code of pvlib.location and found that in version 0.10.3 it defaults to 0 when no altitude parameter is given (Figure 2), but in version 0.11.0 when no altitude parameter is given, it executes a lookup_altitude function (Figure 3), which is implemented to support only a single latitude and longitude inputs, and therefore an error is reported. So if you want to instantiate multiple latitudes and longitudes into a single Location on pvlib version 0.11.0, you need to give altitude, or if not, you need to enter altitude=0 to not report an error (Figure 4).
    Therefore, I think it is necessary to point out this usage on pvilb's official website to facilitate different application scenarios. A screenshot of the code for the batching simulation is attached here (see Figure 5).
[image]

cwhanse, 2024-07-08:
@BYC12138 if this is a duplicate of #2120 please close this issue

kandersolar, 2024-10-19:
duplicate of #2120

closed 2024-10-19 by kandersolar