pvlib/pvlib-python #2125: Proposal: split up mismatch.py

RDaxini, 2024-07-11:
**Is your feature request related to a problem? Please describe.**
I've been working with the spectrum unit of pvlib a fair bit recently for my GSoC project and I find that the [`mismatch.py`](https://github.com/pvlib/pvlib-python/blob/main/pvlib/spectrum/mismatch.py) file currently contains a range of functions, several of which are not directly related to spectral mismatch.

Current functions in `mismatch.py`:

[pvlib.spectrum.get_example_spectral_response](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.get_example_spectral_response.html)
[pvlib.spectrum.get_am15g](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.get_am15g.html) (deprecated v0.11, removed 0.12)
[pvlib.spectrum.get_reference_spectra](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.get_reference_spectra.html)
[pvlib.spectrum.calc_spectral_mismatch_field](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.calc_spectral_mismatch_field.html)
[pvlib.spectrum.spectral_factor_caballero](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.spectral_factor_caballero.html)
[pvlib.spectrum.spectral_factor_firstsolar](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.spectral_factor_firstsolar.html)
[pvlib.spectrum.spectral_factor_sapm](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.spectral_factor_sapm.html)
[pvlib.spectrum.spectral_factor_pvspec](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.spectral_factor_pvspec.html)
[pvlib.spectrum.spectral_factor_jrc](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.spectral_factor_jrc.html)
[pvlib.spectrum.sr_to_qe](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.sr_to_qe.html)
[pvlib.spectrum.qe_to_sr](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.qe_to_sr.html)

**Describe the solution you'd like**
I am wondering whether it might help organise the functions more clearly if we were to split up this file.

One suggestion:

1. `mismatch.py` 
--> covers the effect on PV performance; calculations of spectral mismatch (atmospheric, field, indoor)

2. `spectral_irr.py`
--> covers the outdoor meteorological side of things; all spectral irradiance calcs, e.g. `get_reference_spectra`, possibly move `spectrl2` into here as well

4. `spectral_resp.py` 
--> covers the PV characteristics side of things, e.g. sr_to_qe / qe_to_sr calculations, `get_example_spectral_response`

I think this framework fits with what we have currently and, in terms of future development, I see scope for expansion of `spectral_irr.py` in particular to include for example, function to calculate spectral indices such as the average photon energy. I can see `mismatch.py` being expanded to include variants of the standard spectral mismatch factor, or alternatives such as the (weighted) useful fraction.

**Describe alternatives you've considered**
Could just rename `mismatch.py` entirely to broaden its scope, but I think creating a few files here instead would be better for organisation and future development.

**Additional context**
The suggestion above is just one idea to illustrate the what I mean. I am keen to hear other views on this proposal.

1. Is splitting `mismatch.py` a good idea?
If yes:
2. Alternative categorisation structure?
3. Alternative file names?

I'd be happy to work on this if we reach a consensus. I'm looking forward to hearing your views.

mikofski, 2024-07-11:
You mean `spectrum.py`? Why not make it a sub-package?
- `pvlib/spectrum/mismatch.py`
- `pvlib/spectrum/irradiance.py`
- Etc

I’m -1 on using a pvlib sub package above spectrum called mismatch because it has many meanings, like electrical mismatch or array level mismatch etc

another idea instead of mismatch use`pvlib.spectral.shift`? “Spectral shift” is also sometimes used

RDaxini, 2024-07-11:
> I’m -1 on using a pvlib sub package above spectrum called mismatch because it has many meanings, like electrical mismatch or array level mismatch etc

@mikofski I don't mean to create a package above spectrum called mismatch. I agree that would not be a good idea. I mean splitting up the existing `mismatch.py` into several subpackages, all definitely still within `spectrum`.

> * `pvlib/spectrum/mismatch.py`
> * `pvlib/spectrum/irradiance.py`
> * Etc

This is also what I meant. So what we have currently is:
* `pvlib/spectrum/mismatch.py`
* `pvlib/spectrum/spectrl2.py`

and what I was thinking about is taking some functions out of `mismatch.py` and putting them into some new subpackages (possible new subpackage names in the original post) still within within `pvlib\spectrum`

I hope this makes my suggestion clearer than in the original post 😅

RDaxini, 2024-07-15:
Thanks for your suggestions @echedey-ls

> and put groups 2 & 3 into a new `responses.py` file

On this point, what is your thinking behind combining the groups?
I lean more towards separate groups since I think the functions would be sufficiently distinct from one another. Group 2 is based on spectral irradiance data, while group 3 is based on the PV device.  Just for example, I don't think a function to calculate a parameter such as the average photon energy (to characterise a spectral irradiance distribution) would fit in a `responses.py` package, but a spectral response curve or associated response curve analysis functions would.

> Another point, do we need deprecations for this?

Good point. I am not sure --- @AdamRJensen @kandersolar ?

echedey-ls, 2024-07-15:
> On this point, what is your thinking behind combining the groups?

Mainly because the main topic seems that they are related to calculating spectral mismatch or response from an SR and a spectrum. Also, because the prefix `spectral` in `spectral_irr.py` and `spectral_resp.py` seems redundant since those files will be under `pvlib.spectrum`. And from the contractions `irradiance` and `response`, irradiance is somewhat already taken:

```python
from pvlib import irradiance
from pvlib.spectrum import irradiance
```
This wouldn't work if a file is named irradiance.py under spectrum/. But yeah, we could find some other name I guess.

And then we come to the first point, which is why split things that are going to be used for a common purpose.

Anyway, not a strong opinion; I'm grateful that the base code gets broke down into smaller and maintainable pieces.

AdamRJensen, 2024-07-17:
Because the names of these sub-files aren't seen by the user (functions are imported from pvlib.spectrum), then we're more lenient on using longer filenames. Another response for the prefix, as pointed out by @echedey-ls, is that there's already a file named irradiance.py.

Given that the functions are advertised in the documentation as imported from pvlib.spectrum, then we can justify not having a deprecation cycle for the filename changes.

As we discussed during today's GSoC group meeting, then I think the preferred naming is:
- spectral_response.py
- spectral_irradiance.py
- spectral_mismatch.py

Also, don't forget to update the `__init__.py` file.

cwhanse, 2024-07-17:
@AdamRJensen can you write a one sentence description of each of these modules? For those not at the GSoC meeting.

AdamRJensen, 2024-07-17:
> @AdamRJensen can you write a one sentence description of each of these modules? For those not at the GSoC meeting.

@RDaxini could you do that? I'm back in the saddle peddling my way through Greece

RDaxini, 2024-07-17:
So in the GSoC meeting we talked more specifically about the naming and combining subpackages or not. Since the names are similar to the original post suggestion, my understanding is the content is broadly similar too:

* `pvlib/spectrum/spectral_mismatch.py`
--> covers the spectral effect on PV performance, ie models to calculate spectral mismatch
* `pvlib/spectrum/spectral_irradiance.py`
--> covers the meteorological ('off/non-PV') aspect, ie spectral irradiance calculations such as get_reference_spectra, or new functions such as get_ape (average photon energy)
* `pvlib/spectrum/spectral_response.py`
--> covers the PV side of things, e.g. sr_to_qe/qe_to_sr calculations, get_example_spectral_response

ps welcome to join our fun meetings next time @cwhanse :D

cwhanse, 2024-07-17:
Thanks, that clarifies for me the difference between `spectral_mismatch` and `spectral_response`.

I'll let you know if I need another meeting :)

wholmgren, 2024-07-18:
I recommend against repeating "spectral" in the module name - they're already in the `spectrum` namespace! There's no conflict unless a user tries to import in the way that @echedey-ls demonstrated and the answer for that user is "don't do it that way, instead do something like `import pvlib.spectrum.irradiance as spectral_irradiance`". And don't forget that many of the function names have "spectral" in them too.

RDaxini, 2024-07-19:
Should we vote?
Thumbs up this comment if you're **for** the `spectral_` prefix, thumbs down this comment if you're **against** the prefix?

AdamRJensen, 2024-07-21:
> Should we vote? Thumbs up this comment if you're **for** the `spectral_` prefix, thumbs down this comment if you're **against** the prefix?

@RDaxini I think we can conclude that there is a favor for nixing the prefix, you can update #2136 accordingly

RDaxini, 2024-07-21:
@AdamRJensen ok will do. No computer and limited Internet atm, will do it when I'm back at the end of next week.

PR #2136 (2024-08-05): restructure pvlib/spectrum
 - [X] Closes #2125
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [X] New code is fully documented. In [... cut]
closed 2024-08-05 by kandersolar